carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #3: Using this as a PlatformIO library

MisterSandman7, 2021-05-28:
I'm sorry if this is not the correct place to ask but I can't find a proper resource on how to "convert" this library to a plarformio one.

carlk3, 2021-05-31:
Interesting idea. I haven't used PlatformIO yet, but I'll look into it.

Shiba-Kar, 2022-04-01:
> I'm sorry if this is not the correct place to ask but I can't find a proper resource on how to "convert" this library to a plarformio one.

Have you imported this library into Platform IO. if so can you guide me.

Shiba-Kar, 2022-04-01:
> Interesting idea. I haven't used PlatformIO yet, but I'll look into it.

This can help you

https://docs.platformio.org/en/latest/librarymanager/creating.html#:~:text=PlatformIO%20has%20a%20built%2Din,json%2C%20library.

carlk3, 2022-04-05:
Unfortunately, I don't have time to look into it now. I've never even used PlatformIO, so I think it would take me a while. However, this is an open project. If someone want to take it on, anyone is welcome to submit a pull request.

carlk3, 2023-02-11:
I have never used Arduino or PlatformIO before, so it was a bit of a struggle, but I have taken a stab at it:

https://registry.platformio.org/libraries/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico

Pull Request #48