carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #6: Question about using the library

thaumatichthys, 2021-07-09:
I think I might be doing something wrong, as the example program does not seem to respond to serial commands, and if I try to use the library in my own program, the pico will hang on f_open.

carlk3, 2021-07-09:
> the example program does not seem to respond to serial commands
You might need to adjust the CMake directives that direct output to UART or USB CDC. (See [Getting started with Raspberry Pi Pico](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf).)

> if I try to use the library in my own program, the pico will hang on f_open
Can you run the debugger (GDB)? Could you post a stack traceback of the hang?

thaumatichthys, 2021-07-10:
I have tried changing the output to both UART and USB, and it did not seem to do much.
Sorry I am quite new to this stuff, I don't know if this is what you mean, but this is what was in the debug console after running f_open in the debugger: 
`Thread 
1 received signal SIGTRAP, Trace/breakpoint trap.
my_assert_func (file=file@entry=0x1000acbc "C:\\rpi pico\\project0\\no-OS-FatFS-SD-SPI-RPi-Pico\\FatFs_SPI\\sd_driver\\sd_card.c", line=line@entry=789, func=func@entry=0x1000b198 <__func__.2> "sd_init_card", pred=pred@entry=0x1000b100 "driver_initialized") at C:\rpi pico\project0\no-OS-FatFS-SD-SPI-RPi-Pico\FatFs_SPI\src\my_debug.c:23
23	        __asm("bkpt #0");
`

analoq, 2021-07-11:
I just ran into this, if I have both USB & UART enabled the demo just hangs.  For my setup it works if I enable USB only and leave UART disabled.

thaumatichthys, 2021-07-11:
How did you compile the example? I'm not sure what I am doing wrong but it always hangs for me, and I have the UART disabled.
How I compiled it:
`cd example
mkdir build
cmake -G "MinGW Makefiles" ..
make`

carlk3, 2021-07-12:
> I have tried changing the output to both UART and USB, and it did not seem to do much.
> Sorry I am quite new to this stuff, I don't know if this is what you mean, but this is what was in the debug [...]

That's saying that a run-time assertion failed at line 789 of sd_card.c:

```
int sd_init_card(sd_card_t *this) {
    TRACE_PRINTF("> %s\n", __FUNCTION__);
    myASSERT(driver_initialized);
...
```
`driver_initialized` is a flag that gets set in `sd_init_driver()`. 

Is there a call to `sd_init_driver` before your call to `sd_init_card` before your call to `f_open`?

carlk3, 2021-07-12:
> I have tried changing the output to both UART and USB, and it did not seem to do much.
> Sorry I am quite new to this stuff, I don't know if this is what you mean, but this is what was in the debug [...]

By "stack traceback" I mean what Visual Studio calls "CALL STACK":
[image]

In the GDB command line interface, this is the `where` command.

thaumatichthys, 2021-07-13:
Ok, it seems that I have forgotten to include the sd_init_driver call. After including this, everything seems to work fine. Sorry for wasting your time!

closed 2021-07-13 by thaumatichthys

ericinventor, 2021-07-20:
I am having the same issue where the example code compiles fine but when I load it onto the pico the serial hangs up and I can't access the help menu or run any commands. I only have the USB output enabled since I am programming over usb. Please let me know how to fix this.

thaumatichthys, 2021-07-20:
I have not tried to get the example to work, but this should work:

`
//main.cpp
#include "stdio.h"
#include "hardware/adc.h"
#include "hardware/dma.h"
#include "ff.h"

extern "C"
{
#include "sd_card.h"
}

FIL fil;
FATFS fatfs;

bool sd_ok = 0;

int main()
{
  sd_init_driver(); //important

  FRESULT fr = f_mount(&fatfs, "", 1);
  if (fr == FR_OK)
  {
    sd_ok = 1;
  }

  f_open(
      &fil,
      "text.txt",
      FA_OPEN_APPEND | FA_WRITE);

  f_printf(&fil, "asdf");
  f_close(&fil);
}`

`
//CMakeLists.txt
cmake_minimum_required(VERSION 3.12)
include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)

project(project0 C CXX ASM)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

pico_sdk_init()

add_executable(${PROJECT_NAME}
    main.cpp
)

pico_add_extra_outputs(${PROJECT_NAME})

target_link_libraries(${PROJECT_NAME}
    pico_stdlib
    hardware_adc
    hardware_dma
    FatFs_SPI
)

add_subdirectory(
    no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI build
)

# pico_enable_stdio_usb(${PROJECT_NAME} 0)
pico_enable_stdio_uart(${PROJECT_NAME} 1)

IF (NOT DEFINED N_SD_CARDS)
    SET(N_SD_CARDS 1)
ENDIF()`

and the library folder is in the same directory as main.cpp

ericinventor, 2021-07-24:
Hi @thaumatichthys @carlk3 . Thanks for you help. So I tried compiling the code you have above. First I was getting compile issues complaining about the "*this" used in many of the functions in sd_card.h, which after some research I realized is why you included that file in an extern "C", but it still wasnt working so I just changed my main file to main.c and compiled it as a C project, which worked fine. I assume your example code is supposed to create a file called text.txt with "asdf" written in it and  store it on the sd card. So the good news is my sd card breakout has an activity LED that lights up whenever the card is being written to or read from. So whenever I restart the Pico the activity LED does blink for about a second. But whenever I check the SD card, it shows as totally blank, no files on it whatsoever. I have tried formatting the SD card with FAT32 and exFAT, but neither worked. I'm using adafruits SD card breakout and a Gigastone 8gb class 10 sd card. Please let me know what I may be doing wrong. I appreciate your help! Thanks

ericinventor, 2021-07-24:
OMG!!! I flipped the MOSI and MISO pins and now it is working!!!! Thank you both for your help! @carlk3 please confirm that the table in the readme has been corrected.

carlk3, 2021-07-25:
> OMG!!! I flipped the MOSI and MISO pins and now it is working!!!! Thank you both for your help! @carlk3 please confirm that the table in the readme has been corrected.

I have tried to correct the table twice, now, so I think I need someone else to confirm that it is now correct.

thaumatichthys, 2021-07-25:
@ericinventor Oh yeah sorry about that. In my case I renamed all the instances of *this in the library to something else to get it to work with cpp

carlk3, 2021-08-13:
I have now committed a new version that implicitly calls sd_init_driver(), so you don't need to call it explicitly. 
Also, I have made it C++ compatible.