carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #7: Help compiling library for GCC (C++)

MisterSandman7, 2021-08-07:
The library compiles fine for .c files, but when switching the main to .cpp and updating the Cmake configuration 8 errors pop up from the sd_driver folder, all saying `expected ',' or '...' before 'this'`.
These errors appear in the files `sd_card.h` between lines 64 to 70 and in 'spi.h` between lines 46 to 49.

I'm not sure how to solve this issue (I tried to extern "C", but that didn't do anything).

Any help would be appreciated.
Here's both the log and error files if they can be of any help.
[CMakeError.log](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/files/6949868/CMakeError.log)
[CMakeOutput.log](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/files/6949871/CMakeOutput.log)

carlk3, 2021-08-09:
I shouldn't have used the identifier "this", since that is a keyword in C++. For now, I think you could just do a global search and replace of "this" with "me", or "inst", or whatever. But I will correct this.

MisterSandman7, 2021-08-09:
Thought the same, `this` is a C++ keyword. I'll try to patch up and see if it compiles.

carlk3, 2021-08-13:
I have committed a new version that works with a C++ main.

closed 2021-08-13 by carlk3