carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #8: Does not work when using with scanvideo from pico-extras

jfoucher, 2021-08-15:
This is the pico-extras repository : https://github.com/raspberrypi/pico-extras
When using this project together with scanvideo_dpi, the spi initialization hangs when requesting DMA channels.

Specifically [these two lines](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/73ed88abb2b5f3d8368349a7a51d73338f4f42da/FatFs_SPI/sd_driver/spi.c#L119):
```
    pSPI->tx_dma = dma_claim_unused_channel(true);
    pSPI->rx_dma = dma_claim_unused_channel(true);
```
These function calls never return and the Pico does not do anything else.

Any idea as to where this might be coming from ?
AFAIK the scanvideo code uses a single DMA channel.
The bug occurs as soon as I include scanvideo, I don't even need to call any code from it.

If it helps, scanvideo would be called on the second core.

jfoucher, 2021-08-15:
Ok, I managed to get it working by changing lines 150-154 to this : 

    irq_set_exclusive_handler(DMA_IRQ_1, pSPI->dma_isr);
    // Tell the DMA to raise IRQ line 1 when the channel finishes a block
    dma_channel_set_irq1_enabled(pSPI->rx_dma, true);
    irq_set_enabled(DMA_IRQ_1, true);

Basically just replace DMA IRQ 0 by 1.

I don't know if there is a way to check which dma irq is used already, but if so it might be a way to fix this.

carlk3, 2021-08-16:
I have opened a Pull Request to make the use of DMA_IRQ_0 sharable:
https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/pull/10#issue-972028198

closed 2021-09-14 by carlk3