carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #9: re-initialize SD card after hotplug

wingyukyuk, 2021-08-16:
I am sorry i was new of FATFS and pi pico,

May i know is any method to re-initialize the SD card after hotplug.

I try different way but it still show as below:
```
No response CMD:17
No response CMD:17
No response CMD:17
No response CMD:17 response: 0xff
f_mount error: A hard error occurred in the low level disk I/O layer (1)
```

the code i used is in the example but remove `panic`, because i want my pico still running after unable mount SD card

```
    static FATFS fatfs;
    FRESULT fr = f_mount(&fatfs, "", 1); 
    if (FR_OK != fr)
    {
        printf("f_mount error: %s (%d)\n", FRESULT_str(fr), fr);
        return;
    } 
    FIL fil;
    const char * const filename = "filename.txt";
    fr = f_open(&fil, filename, FA_OPEN_APPEND | FA_WRITE);
    if (FR_OK != fr && FR_EXIST != fr)
    {
        printf("f_open(%s) error: %s (%d)\n", filename, FRESULT_str(fr), fr);
        return;
    } 
    if (f_printf(&fil, "Hello, world!\n") < 0) {
        printf("f_printf failed\n");
    }
    fr = f_close(&fil);
    if (FR_OK != fr) {
        printf("f_close error: %s (%d)\n", FRESULT_str(fr), fr);
    }
    f_unmount("");
```

I try to
```
 memset(&fatfs,0,sizeof(FATFS));
```
```
f_unmount("");
```
```
f_mount("0:");
f_unmount("0:");
```
```
 f_chdrive("");
 f_chdrive("0:");
```
```
FATFS fatfs;
```

But still unable re-initialize SD card. May i know anything i was missing?

Many Thanks ;)

carlk3, 2021-08-16:
Do you have Card Detect hooked up? If so, do you see messages like this:

GPIO 22 interrupt (Card Detect 0:)

when you remove the card (after it is mounted)?

wingyukyuk, 2021-08-17:
I don't have "Card Detect" on my SD card reader. is it possible to re-initialize without hardware detection?

carlk3, 2021-08-17:
You could manually do the same thing that the card detect interrupt does: set `STA_NODISK | STA_NOINIT` bits of the `m_Status` member of the sd_card_t instance. (See `card_detect_callback` in `sd_card.c`.) 

[f_unmount](http://elm-chan.org/fsw/ff/doc/mount.html) does not do that automatically, maybe because FatFs supports multiple partitions on a card (or other physical media). `f_unmount` is part of the Filesystem Layer, but [disk_status](http://elm-chan.org/fsw/ff/doc/dstat.html) is part of the storage device control module. Anyway, you could always make a wrapper around `f_unmount`.

I would highly recommend implementing card detect if you're going to do hot swapping. If you forget to unmount and insert a different card, you're bound to get a corrupted filesystem.

wingyukyuk, 2021-08-17:
understand, and it work! thanks a lot.
And i will buy a card reader with CD function later, thanks for your advice.

closed 2021-08-17 by wingyukyuk