carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #11: Configurable GPIO pins

fruit-bat, 2021-09-02:
Firstly, thanks for publishing this. I am using it my mini-project (https://github.com/fruit-bat/pico-sorcerer-2) and it works really well.

I would love to be able to configure the GPIO pins without having to modify the source code... my sub-module on github won't have the changes needed for other people to download and compile. 

Thanks in advance,

fruit-bat

carlk3, 2021-09-05:
I removed `hw_config.c` from the `FatFs_SPI` library, since I can see how that would interfere with using the library as a submodule. Now, you can define the functions declared in `FatFs_SPI/sd_driver/hw_config.h` wherever and however you'd like. 

I added an example showing how you might set up the configuration at run time.

I have opened a pull request: https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/pull/12#issue-988611933. Please take a look.

fruit-bat, 2021-09-07:
Many thanks for the changes, much appreciated.
I will try them out as soon as I get a  chance and get back to you....

closed 2021-09-14 by carlk3