carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #17: Not really an issue but there seems to be no better way to communicate with author

louigi600, 2021-12-02:
Horry to bother you with this not really issue thing ... but as a radio operator I wanted to add a little note.
Obviously keeping the leeds as short as one can is a good thing but just a not on the limits where one should start to worry about it:
HF band is 3 to 30 MHz and at those sort of frequencies 1/4 wavelength is in the range of 2.5m ro 25m.
It is somewhat less then optimal but I think you would get away with 1/40 wavelength leads ... which in the worst case would be 25cm.
It turns out that the max SCK output frequency the RP2040 can support is 66MHz (already in the VHF band) ... I somehow doubt one will ever go that high unless you want the pico just to do SPI I/O and not much more (let alone on a breadboard) but in that case I think you might still get away with 11cm leads.
Now let's suppose that it is more likely that you will be working with a SCK frequency of 11MHz (wich is still the best the PR2040 can do in input) 1/400 wavelength would be about 7cm so just don't get too worried about the leads being too long.

carlk3, 2021-12-03:
I'm not talking about RFI; just making the thing work. I don't know what it is, but it has happened to me more than once with hastily wired SPI where I have spent hours scratching my head over some intermittent problem, only to find that that problem went away when I rewired with some shorter leads. Maybe some kind of cross coupling? Transmission line effects? I don't know and I certainly don't have the equipment to find out. I just got myself a [SparkFun RP2040 Thing Plus](https://learn.sparkfun.com/tutorials/rp2040-thing-plus-hookup-guide/hardware-overview), with the built-in µSD card slot, which I'm hoping will make life easier (at least until I need to add a second card).

louigi600, 2021-12-03:
How is it working on the STM stuff ?
Anyway I have another issue ... If I try to run the sdcard off the pico's 3.3V out the pico won't even boot.
Have you ever seen this happen ?
I will try with a newer card that should be less power thirsty ...

carlk3, 2021-12-03:
> How is it working on the STM stuff ? 

It is interesting. It is a whole different paradigm than the Pico SDK. A pretty big learning curve for me. But, they make some things really easy. For example, DMA is trivial to implement, at least if you're doing standard things like ADC to memory or SPI to memory. FatFS is actually supplied with the STM32CubeIDE as a "Middlewares" component, but the implementation seems to be incomplete for this low-end chip. FreeRTOS, too, but their implementation makes you use the [CMSIS-RTOS](https://developer.arm.com/tools-and-software/embedded/cmsis) interface.
...
> 
> oooops I realized that my card reader slot salvaged from a broken SD card reader actually has the contact for pad 9 ... was not expecting that since it is not used ... it was a simple wiring problem.
> 
Glad you found it. I power two SD cards from that pin. In theory, it should have enough power, but sometimes I wonder. Adding some decoupling caps. might help.

> PS: my leads are about 25cm long but no parasitic capacitance from breadboard ... the basic write seems to work. I think that on the breadboard the inductance coupled with the relatively high parasitic capacitance plays relay badly for experiments involving high frequency. get rid of the breadboard ans start using something like my castellations socket for pri pico: [thingiverse.com/thing:5140902](https://www.thingiverse.com/thing:5140902)

The castellations socket looks interesting.

I have taken the plunge into [EasyEDA](https://easyeda.com/) and [JLCPCB](https://jlcpcb.com/?href=easyeda-home). Pretty affordable! I'm about ready to send off my first PCB: [Volt Reg Controller Board](https://easyeda.com/carlk3/volt-reg-1-1_copy_copy_copy) (design document: [Automotive Generator Digital Voltage Regulator](https://docs.google.com/document/d/1WW0McwKxeDSn61u2VuPdqIq2SuchMSvakl44oqSDAts/edit?usp=sharing)). Once I have taken a complete path through this process, I hope I will find it cheap and easy to make custom PCBs. The shipping is really the biggest expense and delay.

closed 2021-12-11 by carlk3

louigi600, 2022-03-13:
Interesting thing the Vreg controller board.
I wonder if it is possible to make a switching DC-DC converter to power Pico or Arduino from a e-scooter battery pack?  They are  tipically 10s liion (operating voltage 42v down to around 30v) but I plan to hack it and add a few cells.
I would really like to avoid linear regulator because it's going to be horribly inefficient to drop from 40v down to 5v.