carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #25: Fails to compile with SDK 1.2.0 and 1.3.0

bablokb, 2022-01-30:
Just a note to document this problem and a workaround:

The project fails to compile with pico-sdk 1.2.0. "mutex_is_initialized" is mispelled in pico/mutex.h as "mutex_is_initialzed".

The 1.3.0 sdk fixed this error, but there is a bigger regression which prevents compilation. That is fixed in the develop-branch for the next release.

I decided to stick with 1.2.0 and fix the spelling in pico/mutex.h. After that, I could successfully build the examples.

Python380, 2022-07-28:
SDK 1.4 does not work either.
I have tried to find the origin of the linker errors, but as of yet, have not fully solved the problem.

Python380, 2022-07-28:
Created fork: [https://github.com/Python380/no-OS-FatFS-SD-SPI-RPi-Pico](https://github.com/Python380/no-OS-FatFS-SD-SPI-RPi-Pico)

Python380, 2022-07-28:
Current errors (Using NMake on Windows 11):
```
C:\Users\henri\pico\pico_sd\build>nmake

Microsoft (R) Program Maintenance Utility Version 14.30.30706.0
Copyright (C) Microsoft Corporation.  All rights reserved.

PICO_SDK_PATH is C:/Users/henri/pico/pico-sdk
PICO platform is rp2040.
Build type is Release
PICO target board is pico.
Using board configuration from C:/Users/henri/pico/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at C:/Users/henri/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/henri/pico/pico_sd/build
[  1%] Performing build step for 'ELF2UF2Build'

Microsoft (R) Program Maintenance Utility Version 14.30.30706.0
Copyright (C) Microsoft Corporation.  All rights reserved.

[100%] Built target elf2uf2
[  2%] No install step for 'ELF2UF2Build'
[  3%] Completed 'ELF2UF2Build'
[  8%] Built target ELF2UF2Build
Scanning dependencies of target bs2_default
[ 10%] Built target bs2_default
[ 12%] Built target bs2_default_padded_checksummed_asm
Scanning dependencies of target sd_fat_spi
Consolidate compiler generated dependencies of target sd_fat_spi
[ 13%] Building C object CMakeFiles/sd_fat_spi.dir/main.c.obj
[ 14%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c.obj
[ 15%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_gpio/gpio.c.obj
[ 16%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_claim/claim.c.obj
[ 17%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_platform/platform.c.obj
[ 18%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_sync/sync.c.obj
[ 19%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c.obj
[ 20%] Building ASM object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_irq/irq_handler_chain.S.obj
[ 21%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_sync/sem.c.obj
[ 22%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_sync/lock_core.c.obj
[ 23%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_time/time.c.obj
[ 24%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_time/timeout_helper.c.obj
[ 25%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_timer/timer.c.obj
[ 26%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_util/datetime.c.obj
[ 27%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_util/pheap.c.obj
[ 28%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_util/queue.c.obj
[ 29%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_sync/mutex.c.obj
[ 30%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/common/pico_sync/critical_section.c.obj
[ 31%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_uart/uart.c.obj
[ 32%] Building ASM object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_divider/divider.S.obj
[ 33%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c.obj
[ 34%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_clocks/clocks.c.obj
[ 35%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_pll/pll.c.obj
[ 36%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_vreg/vreg.c.obj
[ 37%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_watchdog/watchdog.c.obj
[ 38%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/hardware_xosc/xosc.c.obj
[ 39%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_printf/printf.c.obj
[ 40%] Building ASM object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_bit_ops/bit_ops_aeabi.S.obj
[ 41%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_bootrom/bootrom.c.obj
[ 42%] Building ASM object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_divider/divider.S.obj
[ 43%] Building ASM object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_double/double_aeabi.S.obj
[ 44%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_double/double_init_rom.c.obj
[ 45%] Building C object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_double/double_math.c.obj
[ 46%] Building ASM object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_double/double_v1_rom_shim.S.obj
[ 47%] Building ASM object CMakeFiles/sd_fat_spi.dir/C_/Users/henri/pico/pico-sdk/src/rp2_common/pico_int64_ops/pico_int64_ops_aeabi.S.obj
[... cut ...]
glue.c:(.text.disk_ioctl+0x6): undefined reference to `sd_get_by_num'
collect2.exe: error: ld returned 1 exit status
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR10B2~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.30.30705\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.30.30705\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
```

Python380, 2022-07-28:
Changing (`FatFs_SPI/CMakeLists.txt`)
```cmake
add_library(FatFs_SPI INTERFACE)
target_sources(FatFs_SPI INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}/ff14a/source/ffsystem.c
    ${CMAKE_CURRENT_LIST_DIR}/ff14a/source/ffunicode.c
    ${CMAKE_CURRENT_LIST_DIR}/ff14a/source/ff.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/sd_spi.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/demo_logging.c
#    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/hw_config.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/spi.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/sd_card.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/crc.c
    ${CMAKE_CURRENT_LIST_DIR}/src/glue.c
    ${CMAKE_CURRENT_LIST_DIR}/src/f_util.c
    ${CMAKE_CURRENT_LIST_DIR}/src/ff_stdio.c
    ${CMAKE_CURRENT_LIST_DIR}/src/my_debug.c
    ${CMAKE_CURRENT_LIST_DIR}/src/rtc.c
)
target_include_directories(FatFs_SPI INTERFACE
    ff14a/source
    sd_driver
    include
)
target_link_libraries(FatFs_SPI INTERFACE
        hardware_spi
        hardware_dma
        hardware_rtc
        pico_stdlib
)
```
to
```cmake
add_library(FatFs_SPI INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}/ff14a/source/ffsystem.c
    ${CMAKE_CURRENT_LIST_DIR}/ff14a/source/ffunicode.c
    ${CMAKE_CURRENT_LIST_DIR}/ff14a/source/ff.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/sd_spi.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/demo_logging.c
#    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/hw_config.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/spi.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/sd_card.c
    ${CMAKE_CURRENT_LIST_DIR}/sd_driver/crc.c
    ${CMAKE_CURRENT_LIST_DIR}/src/glue.c
    ${CMAKE_CURRENT_LIST_DIR}/src/f_util.c
    ${CMAKE_CURRENT_LIST_DIR}/src/ff_stdio.c
    ${CMAKE_CURRENT_LIST_DIR}/src/my_debug.c
    ${CMAKE_CURRENT_LIST_DIR}/src/rtc.c
)
target_include_directories(FatFs_SPI INTERFACE
    ff14a/source
    sd_driver
    include
)
target_link_libraries(FatFs_SPI INTERFACE
        hardware_spi
        hardware_dma
        hardware_rtc
        pico_stdlib
)
```
fails with the following:
```
[ 15%] Linking CXX executable sd_fat_spi.elf
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/sd_fat_spi.dir/main.c.obj: in function `main':
main.c:(.text.startup.main+0x36): undefined reference to `sd_init_driver'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0x4e): undefined reference to `f_mount'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0x64): undefined reference to `f_open'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0x70): undefined reference to `f_printf'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0x7c): undefined reference to `f_printf'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0x86): undefined reference to `f_close'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0xa6): undefined reference to `f_close'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0xb4): undefined reference to `f_close'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0xc0): undefined reference to `f_open'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0xee): undefined reference to `f_gets'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0xfe): undefined reference to `f_close'
c:/progra~2/gnuarm~1/102020~1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: main.c:(.text.startup.main+0x116): undefined reference to `f_mount'
collect2.exe: error: ld returned 1 exit status
NMAKE : fatal error U1077: 'C:\PROGRA~2\GNUARM~1\102020~1\bin\AR10B2~1.EXE' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.30.30705\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Tools\MSVC\14.30.30705\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
```

Python380, 2022-07-28:
reverting change to makefile & adding a hw_config.c in the `sd_driver` directory (and uncommenting commented line in makefile) results in the same error.

Python380, 2022-07-28:
This is where I get stumped... certainly seems like `sd_init_driver` is defined to me...

Python380, 2022-07-28:
Did not update to pico-sdk 1.4.0 properly.
Tried to remove, reclone, and initialize submodules.
Now the only issue was that the way the lib is structured, it requires you to copy the hw_config.c from the example directory in the repo to your project or `FatFs_SPI/sd_driver` and updating the relevant cmake files...

Have not tested properly, but that gets it to build.

TheMobas0, 2022-07-28:
> Did not update to pico-sdk 1.4.0 properly. Tried to remove, reclone, and initialize submodules. Now the only issue was that the way the lib is structured, it requires you to copy the hw_config.c [...]

Worked for me, thanks

carlk3, 2022-08-15:
I probably should have documented this somewhere, but [Pull Request #12 Dynamic configuration](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/pull/12) (in response to [Issue #11 Configurable GPIO pins](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/issues/11)) moved the `hw_config.c` file out of the library itself and into the examples, so if you are migrating an application from a version prior to [commit 2260521d602b8427fb779275cf51af6702312ac2](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/pull/12/commits/2260521d602b8427fb779275cf51af6702312ac2) (Sep 5, 2021), you will need to grab a copy from one of the examples (and customize it if necessary), or write some other implementation (e.g., `dynamic_config_example/hw_config.cpp`), and put it somewhere persistent.

I have just pulled SDK version 1.4.0 and no-OS-FatFS-SD-SPI-RPi-Pico/example builds cleanly with that.

closed 2022-08-30 by carlk3