carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #26: CMD:13

philatkin, 2022-02-27:
I see strange behaviours on certain SD cards that II don't know how to diagnose. Frequently 

No response CMD:13

then everything appears to work fine and files are written correctly. 

But on a 64GB Integral SD card I see this repeated message sequence :

R2: 0x33ff
Card is Locked                         
WP Erase Skip, Lock/Unlock Cmd Failed  
Error                                  
CC Error                               
Card ECC Failed                        
WP Violation                           
Erase Param                            
Out of Range, CSD_Overwrite            
In Idle State                          
Erase Reset                            
Erase Sequence Error                   
Address Error             

but files seem to be written successfully.

carlk3, 2022-02-27:
I have spent a lot of time trying to diagnose problems like that only to find that they went away when I moved to a proto-board and soldered everything. Or better yet, a PCB (https://jlcpcb.com/). Also, driving the SD card directly with the GPIOs is not ideal. Take a look at the CM1624 (https://www.onsemi.com/pdf/datasheet/cm1624-d.pdf). Unfortunately, it's a tiny little surface mount part -- not so easy to work with, but the schematic in the data sheet is still instructive. At a minimum, it's probably not a bad idea to have 40 - 100 Ω series resistors at the SD card end of CS, SCK, MISO, MOSI. Pull ups on CS and MISO, at least, are strongly recommended. And, a decoupling cap. or two (e.g., 10 - 100 nF) between 3.3 V and GND on the SD card. 

Besides all that, you can try playing with the baud rate in the hardware configuration. Also, for around $10, something like this Comidox 1Set USB Logic Analyzer Device Set USB Cable 24MHz 8CH 24MHz 8 Channel UART IIC SPI Debug for Arduino ARM FPGA M100 Hot (https://smile.amazon.com/gp/product/B07KW445DJ/) and PulseView - sigrok (https://sigrok.org/wiki/Main_Page) make a nice combination for looking at SPI, as long as you don't run the baud rate too high. But, the last time I tried to diagnose a "No response CMD:13" all that I found out was that yes, the SD card was not sending a response.

philatkin, 2022-02-28:
Thanks for the prompt response. I am using a baseboard PCB that I designed myself, very short traces between Pico and SD module, and the SD module is an Adafruit board mounted to my PCB, and that has integral pullups and a decoupler or two. I don’t have any termination though, so there might be reflections causing issues. I can back off the baud rate somewhat but I am capturing ultrasonic signals, 384kHz mono 16, so I need to sustain “big_file_test” performance of some 800kBytes/sec. Will let you know if and how things improve or not!

carlk3, 2022-02-28:
That looks like a good design. I am struggling to get an SD card working in an automotive environment which is incredibly electromagnetically noisy. (https://www.electro-tech-online.com/threads/automotive-generator-universal-digital-voltage-regulator.159967/page-6#post-1418432) I used the simple SparkFun SD Breakout board, but that didn't work too well. It has no ground plane at all, and seems to act as an antenna. I am now working on a four layer board, with the SD card socket directly on the board. (https://oshwlab.com/carlk3/volt-reg-controller-board-2) I would have liked to use that CM1624, but there is a part shortage at JLCPCB, and that's one I'd rather not have to mount myself.

Anyway, that error message looks like you are getting every possible error from a CMD13_SEND_STATUS response. In other words, the card is sending 0xF, or just holding the line high. Maybe that card simply does not implement the command, but that doesn't seem conformant. You could try commenting out the line:
`    sd_cmd(pSD, CMD13_SEND_STATUS, 0, false, &stat);`
in 
```
static int in_sd_write_blocks(sd_card_t *pSD, const uint8_t *buffer,
                              uint64_t ulSectorNumber, uint32_t blockCnt)
```
in sd_card.c.

The reason that it is there is that 
> SD Specifications
> Part 1
> Physical Layer
> Simplified Specification
> Version 6.00
> April 10, 2017

says, in section _7.2.4 Data Write_:

> After a data block has been received, the card will respond with a data-response token. If the data block
> has been received without errors, it will be programmed. As long as the card is busy programming, a
> continuous stream of busy tokens will be sent to the host (effectively holding the DataOut line low).
> Once the programming operation is completed, the host should check the results of the programming
> using the SEND_STATUS command (CMD13). Some errors (e.g. address out of range, write protect
> violation etc.) are detected during programming only. The only validation check performed on the data
> block, and communicated to the host via the data-response token, is the CRC and general Write Error
> indication.

This doesn't look look like an electrical problem, but a protocol problem. Maybe you could reach out to Integral?

carlk3, 2022-02-28:
I have a hunch that the Integral might be a card that requires the SPI CS to be deasserted (unasserted?) between commands. The _Specifications_ says only that "The host starts every bus transaction by asserting the CS signal low." To me, that doesn't say that it must be deasserted high first. But others might have a different interpretation. 

I don't have an Integral card to test with, but if you'd like to try a fix that deasserts CS between the CMD24_WRITE_BLOCK or CMD25_WRITE_MULTIPLE_BLOCK and the CMD13_SEND_STATUS, you can pull it from the "dev" branch (https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/dev).

philatkin, 2022-03-01:
Thanks for all this, I will give it a shot over the next couple of hours.

carlk3, 2022-03-01:
That breakout board looks a lot better than the Sparkfun one. After seeing the amount of vibration involved in mounting something on an engine, I have decided to go with a hinged socket. A company in the UK makes a nice looking one:
https://gct.co/connector/mem2067
but since I am using JLCPCB Assembly on my current project, I went with a Chinese one that they have in stock.

philatkin, 2022-03-01:
I surface-mount the Adafruit to my baseboard (mock up image here), and am reworking the PCB right now to try to improve decoupling and generally increase capacitive charge storage, I get extraordinary amounts on noise injected into the system when the SD card is writing- it makes a horrible racket in the headphones and I’m concerned it’s the Pico not being able supply enough 3v3 to the Adafruit.

https://www.omenie.com/pipistrelle.html

carlk3, 2022-03-02:
> Have you ever tried setting the GPIO "Slew rate control" to "Slow"? 
Never mind. The default appears to be GPIO_SLEW_RATE_SLOW.

carlk3, 2022-03-03:
> I get extraordinary amounts on noise injected into the system when the SD card is writing- it makes a horrible racket in the headphones and I’m concerned it’s the Pico not being able supply enough 3v3 to the Adafruit.

It occurred to me that reducing the GPIO drive strength might help with the noise. I added some options to set GPIO output drive strengths in the hardware configuration. If you want to try it, checkout dev, and add these lines to hw_config:

```
// In static spi_t spis[] = {  // One for each SPI.
        .set_drive_strength = true,
        .mosi_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,
        .sck_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,

// In static sd_card_t sd_cards[] = {  // One for each SD card
        .set_drive_strength = true,
        .ss_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,

```
The default appears to be GPIO_DRIVE_STRENGTH_4MA. 2MA seems to work fine with a SanDisk 16GB card in a Thing Plus at baud rate 20,833,333.

closed 2023-07-28 by carlk3