carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #28: Cannot mount sd card

oguzkaganyaglioglu, 2022-04-30:
Hello,
I am trying to use this library. But I cannot mount sd card. For example, I uploaded the FatFS_SPI_example.cpp example then I send `mount 0:` command. After that I receive that error:

```
> mount 0:
sd_spi_go_low_frequency: Actual frequency: 398089
V2-Version Card
R3/R7: 0x1aa
Illegal command CMD:59 response 0x5
R3/R7: 0x80ff8000
R3/R7: 0x80ff8000
Card Initialized: Standard Capacity Card: Version 2.x
SD card initialized
_read_bytes: Invalid CRC received 0xc904 result of computation 0x8657
Couldn't read csd response from disk
f_mount error: The physical drive cannot work (3)
```

I am using this sd module:
[image]

I set my card as Fat32

and I changed sd_cards section in hw_config.c to that
```
static sd_card_t sd_cards[] = {  // One for each SD card
    {
        .pcName = "0:",           // Name used to mount device
        .spi = &spis[0],          // Pointer to the SPI driving this card
        .ss_gpio = 17,            // The SPI slave select GPIO for this SD card

        .use_card_detect = false,
        .card_detect_gpio = 22,   // Card detect
        .card_detected_true = -1,  // What the GPIO read returns when a card is
                                  // present. Use -1 if there is no card detect.
        .m_Status = STA_NOINIT,
    }
};
```

What am I doing wrong?

carlk3, 2022-04-30:
I tried an sd module like that once, and I never could get it to work. I came to the conclusion that it was really meant for 5V (compatible with things like Arduino). You could try disconnecting the VCC pin and sticking a 5 V power supply across it and the GND pin.  (I believe the LDO voltage regulator on that module requires a power supply of 4.5V ~ 5.5V. A few batteries might do it.) Or, you could try hacking the module to bypass the LDO voltage regulator. Somehow, you want to get 3.3V on the SD card's VCC.

My new favorite module for the Pico is the [Adafruit Micro SD SPI or SDIO Card Breakout Board - 3V ONLY!](https://www.adafruit.com/product/4682). At $2.95 in quantity one, it doesn't cost that much more than the Arduino compatible cards if you really only need one or two.

However, if you want to figure this out, I would start by running the Low Level I/O Driver Test with the command "lliot". Much easier to debug at that lower level than with something like "mount".  You can use the debugger to step through the code and see where things go wrong. If it's a hardware problem, you'll probably need something like this: [Comidox 1Set USB Logic Analyzer Device Set USB Cable 24MHz 8CH 24MHz 8 Channel UART IIC SPI Debug for Arduino ARM FPGA M100 Hot](https://smile.amazon.com/gp/product/B07KW445DJ/) and [PulseView - sigrok](https://sigrok.org/) to see what's going on.

carlk3, 2022-04-30:
The LDO regulator looks like it could be something like this: https://www.onsemi.com/pdf/datasheet/ncp1117-d.pdf
(but I can't make out the markings in the photo).
The output on the heatsink tab is probably going to the 3.3V VCC on the SD card socket, with a couple of decoupling capacitors there off to the side. You could try desoldering that chip, or just cut the pins, then run a jumper wire from the pin marked VCC on the board edge connector over to where the heatsink tab on the LDO is. 

Actually, you probably don't even need to remove the LDO. Just run the jumper, effectively shorting the input and output of the regulator, which shouldn't hurt anything. You might find that the bottom of the board is a 3.3V plane, so it might be easier (or neater) to jumper there.

Then you should be able to run on 3.3V.  The other chip appears to be 
https://www.ti.com/lit/ds/symlink/sn74lvc125a.pdf
which should be fine.

carlk3, 2022-05-03:
I just remembered that if you power your Pico by USB, you've already got 5 V sitting on VBUS. So, the simplest thing might be to just power this module at 5 V (as it was designed to be)  by running a line from Pin 40 (VBUS) on the Pico to the pin marked VCC on the micro SD card module.

oguzkaganyaglioglu, 2022-05-03:
Thanks for your help. I tried 5V, I don't know why but it didn't work. So I decided to build Adafruit's 3V ONLY card. And it worked. 

Here is what I got

[image]

closed 2022-05-03 by oguzkaganyaglioglu