carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #29: undefined reference to `sd_get_num' and undefined reference to `spi_get_by_num'

TheMobas0, 2022-07-28:
Can't compile code.

Error message:
```
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/Test.dir/lib/no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI/sd_driver/sd_card.c.obj: in function `sd_init_driver':
sd_card.c:(.text.sd_init_driver+0x42): undefined reference to `sd_get_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: sd_card.c:(.text.sd_init_driver+0x4c): undefined reference to `sd_get_by_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: sd_card.c:(.text.sd_init_driver+0x88): undefined reference to `spi_get_by_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: sd_card.c:(.text.sd_init_driver+0x96): undefined reference to `spi_get_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/Test.dir/lib/no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI/src/glue.c.obj: in function `disk_status':
glue.c:(.text.disk_status+0x2): undefined reference to `sd_get_by_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/Test.dir/lib/no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI/src/glue.c.obj: in function `disk_initialize':
glue.c:(.text.disk_initialize+0x2): undefined reference to `sd_get_by_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/Test.dir/lib/no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI/src/glue.c.obj: in function `disk_read':
glue.c:(.text.disk_read+0xa): undefined reference to `sd_get_by_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/Test.dir/lib/no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI/src/glue.c.obj: in function `disk_write':
glue.c:(.text.disk_write+0xa): undefined reference to `sd_get_by_num'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/Test.dir/lib/no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI/src/glue.c.obj: in function `disk_ioctl':
glue.c:(.text.disk_ioctl+0x6): undefined reference to `sd_get_by_num'
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/Test.dir/build.make:912: Test.elf] Error 1
make[2]: *** [CMakeFiles/Makefile2:1580: CMakeFiles/Test.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:1587: CMakeFiles/Test.dir/rule] Error 2
make: *** [Makefile:131: Test] Error 2
```

Python380, 2022-07-28:
related to #25...

carlk3, 2022-08-15:
Which application were you building? E.g., `example` or `simple_example`? In any case, there should be a customized file that provides those functions. For `example` or `simple_example` it's `hw_config.c`.  For `dynamic_config_example` it's `hw_config.cpp`. The file name is arbitrary (except that it must agree with the `CMakeLists.txt`), but something needs to provide those functions. 

Note that there is no CMakeLists.txt to build anything in the top level `no-OS-FatFS-SD-SPI-RPi-Pico` directory.

I have been using some new hardware and it's hard to keep the various `hw_config`s straight (let alone tested). Don't count on the hw_config in the examples on GitHub to be correct for your hardware (even if it once was). Once you have customized it, keep a copy somewhere, give it a new name, or something like that.

carlk3, 2022-08-15:
One more thing: [Pull Request #12 Dynamic configuration](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/pull/12) (in response to [Issue #11 Configurable GPIO pins](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/issues/11)) moved the `hw_config.c` file out of the library itself and into the examples, so if you are migrating from a version prior to 
[commit 2260521d602b8427fb779275cf51af6702312ac2](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/pull/12/commits/2260521d602b8427fb779275cf51af6702312ac2) (Sep 5, 2021), you will need to grab a copy from one of the examples (and customize it if necessary), or write some other implementation (e.g., `hw_config.cpp`), and put it somewhere persistent.

closed 2022-08-30 by carlk3