carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #32: Card eject-insert in example

B-HWD, 2022-08-22:
To reproduce:
Reset->mount 0:
```
> mount 0:
sd_spi_go_low_frequency: Actual frequency: 398089
Illegal command CMD:8 response 0x5
R3/R7: 0xff8000
Card Initialized: Version 1.x Card
SD card initialized
Standard Capacity: c_size: 3907
Sectors: 0xf4400 : 1000448
Capacity: 0x1e880000 : 488 MB
sd_spi_go_high_frequency: Actual frequency: 20833333

>
```
type "ls" to test IO of card
```
> ls
Directory Listing: 0:/
test1 [directory] [size=0]
numbers.txt [writable file] [size=61]
System Volume Information [directory] [size=0]
cdef [directory] [size=0]
bf [writable file] [size=104857600]

>
```
type "unmount 0:"
```
> unmount 0:

>
```
physically remove card
insert card
type "mount 0:"
```
> mount 0:
No response CMD:17
No response CMD:17
No response CMD:17
No response CMD:17 response: 0xff
f_mount error: A hard error occurred in the low level disk I/O layer (1)

>
```
This is it.

carlk3, 2022-08-24:
I guess you are not using Card Detect. Am I right?

B-HWD, 2022-08-24:
Yes.

carlk3, 2022-08-24:
Please review [Pull Request #34](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/pull/34), and test by pulling from branch `dev`.

B-HWD, 2022-08-24:
Still not working :(
`
// connected to usb-serial port after reset, sd-card inserted before reset
>
> mount 0:
sd_spi_go_low_frequency: Actual frequency: 398089
Illegal command CMD:8 response 0x5
R3/R7: 0xff8000
Card Initialized: Version 1.x Card
SD card initialized
Standard Capacity: c_size: 3907
Sectors: 0xf4400 : 1000448
Capacity: 0x1e880000 : 488 MB
sd_spi_go_high_frequency: Actual frequency: 20833333

> ls
Directory Listing: 0:/
test1 [directory] [size=0]
numbers.txt [writable file] [size=61]
System Volume Information [directory] [size=0]
cdef [directory] [size=0]
bf [writable file] [size=104857600]
1 [directory] [size=0]
filename.txt [writable file] [size=182]

> unmount 0:

// sd-card removed and inserted here

> mount 0:
No response CMD:17
No response CMD:17
No response CMD:17
No response CMD:17 response: 0xff
f_mount error: A hard error occurred in the low level disk I/O layer (1)

>
`

carlk3, 2022-08-25:
What kind of SD card is this? I see that you're getting an error:
_Illegal command CMD:8 response 0x5_
If it doesn't support CMD8 it must be a very old (V1) card (or something is broken). Version 2.00 and later SD cards shall support CMD8. I do see another message:
_Card Initialized: Version 1.x Card_
I don't think I have any V1 cards to test with.

Anyway, I don't think that explains the problem you're having. Do you have the capability to run a debugger?

closed 2022-11-08 by carlk3