carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #35: Add Flash Target support

DatanoiseTV, 2022-09-23:
The RP2040 Flash can be used as a MSC disk. Combining this with your application would offer flexibilty to exchange files from an on Flash FAT and a PC. Maybe you could include it in your example?

```cpp
#include "hardware/flash.h"

// We use the last 1MB of flash for storage
#define RESERVED_FLASH (1 * 512 * 1024)

// We want 1MB of storage
#define FLASH_SIZE 1024 * 1024

// TODO: Split the caching out of supervisor/shared/external_flash so we can use it.
#define SECTOR_SIZE 4096

uint8_t FLASH_disk_read(uint8_t *buff, uint32_t sector, uint8_t count)
{
    memcpy(buff, (void *)(XIP_BASE + RESERVED_FLASH + (sector * 512)), count * 512);
    return 0;
}

uint8_t FLASH_disk_write(uint8_t *buff, uint32_t sector, uint8_t count)
{
    
#define FILESYSTEM_BLOCK_SIZE 2048
  uint32_t blocks_per_sector = SECTOR_SIZE / FILESYSTEM_BLOCK_SIZE;
  uint32_t block = 0;
  uint32_t num_blocks = count;
  while (block < num_blocks)
  {
    uint32_t block_address = sector + block;
    uint32_t sector_offset = block_address / blocks_per_sector * SECTOR_SIZE;
    uint8_t block_offset = block_address % blocks_per_sector;

    if (_cache2_lba != block_address)
    {
      memcpy(_cache2,
             (void *)(XIP_BASE + RESERVED_FLASH + sector_offset),
             SECTOR_SIZE);
      _cache2_lba = sector_offset;
    }
    for (uint8_t b = block_offset; b < blocks_per_sector; b++)
    {
      // Stop copying after the last block.
      if (block >= num_blocks)
      {
        break;
      }
      // point src to buffer
      memcpy(_cache2 + b * FILESYSTEM_BLOCK_SIZE,
             buff + block * FILESYSTEM_BLOCK_SIZE,
             FILESYSTEM_BLOCK_SIZE);
      block++;
    }
    // Make sure we don't have an interrupt while we do flash operations.
    // disable interrupts and store them
    uint32_t ints = save_and_disable_interrupts();
    flash_range_erase(RESERVED_FLASH + sector_offset, SECTOR_SIZE);
    flash_range_program(RESERVED_FLASH + sector_offset, _cache2, SECTOR_SIZE);
    restore_interrupts(ints);
  }
    return 0;
}

```

carlk3, 2022-11-11:
That's an interesting idea. I tried to go the other way and combine a [littlefs](https://github.com/littlefs-project/littlefs) front end with my SPI driver (used in this project). [littlefs](https://github.com/littlefs-project/littlefs) is widely used for flash storage, but it turns out that it does not scale well at all to the size of an SD card.

I would suggest that a FatFS/flash library should probably be its own separate repository. I've already started a separate one for FatFS/SDIO/PIO. It's tempting to make one big framework that allows mixing & matching the various front ends and back ends, but I'm afraid the complexity would drive potential adopters away, and might be a maintenance headache.

DatanoiseTV, 2022-11-12:
I think it would be very very interesting for quickly exchanging data between the computer and a pico. I see a lot of use cases. Afaik, CircuitPython is doing it this way, but when I looked the last time, the implementation was pretty complex.

Would be great to see FAT12/FAT16/FAT32 support, as this is what every OS supports. This would allow drag and drop file transfer.

Btw, after some tryout, I figured out my example corrupts data. You need to align the writes for flash to the corresponding block size, so you need some kind of read/write cache. Maybe this could be some 1K-2K allocated in memory.

carlk3, 2022-11-13:
> I think it would be very very interesting for quickly exchanging data between the computer and a pico. I see a lot of use cases. Afaik, CircuitPython is doing it this way, but when I looked the last time, the implementation was pretty complex.

The use cases are not obvious to me. Could you give some examples?

> This would allow drag and drop file transfer.

Could you elaborate on how that would work? Are you implementing a USB Host Mass Storage Device (MSC)?

DatanoiseTV, 2022-11-13:
@carlk3 For example lets say you are building a datalogger where you are storing the data from sensors in flash in a FATFS. Data transfer to the PC could be just a drag & drop operation in that case. I was using a MSC Class for my tryouts and that is the quickest way to go. This is also interesting to store device settings and changing them from the PC. If you store the settings like ini or json-serialised, it could be changed from the PC.

carlk3, 2022-11-14:
> For example lets say you are building a datalogger where you are storing the data from sensors in flash in a FATFS.

Well, you're not going to be able to log a whole lot of data in flash.

> Data transfer to the PC could be just a drag & drop operation in that case.

Could you elaborate on how that would work? Are you implementing a USB Host Mass Storage Device (MSC)?

DatanoiseTV, 2022-11-14:
@carlk3 Well, with 16MB of flash it is plenty for sensor readings (e.g. humidity logger, each 1 minute, or gps logger activated by motion).

I mean USB MSC Device, so it presents itself as a disk drive to the computer. Like the Pico Bootloader.

carlk3, 2022-11-14:
> Well, with 16MB of flash it is plenty for sensor readings (e.g. humidity logger, each 1 minute, or gps logger activated by motion).

I'm sure there are applications where that would be sufficient. The industrial data logging applications that I have worked on tend to generate gigs of data, which is why I'm focused on SD card storage. We have found that the fastest and friendliest way to transfer that much data to a PC is to provide a user-accessible SD card that can be popped out and carried to the PC.

> I mean USB MSC Device, so it presents itself as a disk drive to the computer. Like the Pico Bootloader.

Alright, let me see if I understand the architecture you're building:

On the Pico,
1. You have a block device driver for flash (code you posted above).
2. The flash storage area is formatted as a FAT filesystem.
3. You have set up TinyUSB as an USB MSC endpoint. It uses the block device driver interface to access the flash. Windows (or whatever's on the other endpoint of the MSC) uses the USB mass storage device as a FAT file system. 
4. You need a FatFS interface that uses the block device driver interface to access the flash. Applications on the Pico will use the FatFS interface to access the flash.

I believe that Windows (or whatever's on the other endpoint of the MSC) will assume exclusive ownership of the block device, so that it can do caching. So, when the USB MSC is mounted, the file system will have to be unmounted by FatFS, so logging (or whatever) will have to be suspended until the USB MSC is unmounted. In other words, you have two potential reader/writers, but only one at a time can access the shared flash resource.

Is my understanding correct?

DatanoiseTV, 2022-11-15:
Yes, correct. And if LBA 0 is written by MSC, then a re-read hook could be fired for FATFS for example.

DatanoiseTV, 2022-11-22:
@carlk3 Yes, that might be worth a separate project. Here is a bit of a similar thing https://www.youtube.com/watch?v=tmmHijhct_k

closed 2023-07-28 by carlk3