carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #38: f_write without advancement of buffer pointer

callous4567, 2022-11-19:
Hello;

Is it possible to make use of f_write (or feasibly modify the code without too much intrusion) to pull data from a data address without having it internally advance the address it pulls from?

Ex:
- Taking data from ADC Fifo address
- Taking data from SPI read/write data register address
- ... 

I'm unsure as to whether your library has modified FatFS extensively (so I'm asking here first! d:) ... I'm also quite new to C/C++ thus you'll have to forgive any erroneous terminology I use... 

If this question isn't suited for the library (and is instead more suited for FatFS itself) please disregard and close this issue- I just figured it was worth a shot! Thanks!

carlk3, 2022-11-19:
I haven't modified FatFS at all (except for the configuration file), but I think you might find what your looking for in the SD/SPI block device driver (what FatFS calls the "storage device control module"). Specifically, have a look at `spi_transfer` in  `sd_driver/spi.c`.

callous4567, 2022-11-20:
Thanks for the hint Carl! Your hint is just what I needed to get me started! I'll report back at some point (I've many other priorities...!) on the results. 

Much obliged! ^_^

closed 2022-11-20 by callous4567

callous4567, 2023-01-18:
Update: I wasn't able to get this working in the end (the SD card didn't write when I just slapped it with the ADC without properly pacing it/etc...) but I managed to get something going to capture the full 499.999 kHz of the Pi Pico ADC...

My quasi-solution is here under "f_write_audiobuf" in case anyone finds this over google 
https://github.com/callous4567/vespertilio/blob/main/SRAM_SD_Testing/drivers/FatFs_SPI/ff14a/source/ff.h

At the moment I've been able to write audio files running at 499.999 kHz @ 16b that are around 1 GiB in size (I haven't tested anything else... realistically I am only using max. 250 mB files anyway.) That's with the SD card clock at 25 MHz & the Pico at the usual 125 MHz. The delay between 250~ish mB files is roughly 1-2 seconds to f_close, initiate a wave header, etc, which is satisfactory enough for me. I hope someone else finds this useful! 

Thanks again for the library @carlk3 it really is fantastic- I don't know which parts you wrote or modified but it's very cleanly written! ^_^

carlk3, 2023-01-18:
Very interesting project! Impressive performance. You've got me thinking again about making a version of this library that uses the 4-wire SDIO protocol instead of SPI, for much greater speed. Of course, that would consume more GPIO pins.

callous4567, 2023-01-23:
> Very interesting project! Impressive performance. You've got me thinking again about making a version of this library that uses the 4-wire SDIO protocol instead of SPI, for much greater speed. Of course, that would consume more GPIO pins.

Aye, SDIO would be fantastic ^_^ - I'm trying to see how far I can get single-bit SPI on this thing but it seems like getting the clock past 33 MHz isn't working... trying at 40 MHz or 50 MHz doesn't seem to get anywhere (I get an Invalid CRC error- I assume I need to enable High Speed Mode but I have no clue.)

carlk3, 2023-01-23:
> I'm trying to see how far I can get single-bit SPI on this thing but it seems like getting the clock past 33 MHz isn't working... trying at 40 MHz or 50 MHz doesn't seem to get anywhere (I get an Invalid CRC error- I assume I need to enable High Speed Mode but I have no clue.)

You might be hitting some electrical limitations. It would be nice if you could put an oscilloscope on the SPI signals. You could experiment with the drive strength. For example, in hw_config:
```
        .set_drive_strength = true,
        .mosi_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,
        .sck_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,

        .ss_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,
```
It can go as high as GPIO_DRIVE_STRENGTH_12MA. 

The line termination can make a big difference. Take a look at the [CM1624](https://www.onsemi.com/pdf/datasheet/cm1624-d.pdf). It is a tiny surface mount device meant to go into phones, I think, but its ELECTRICAL SCHEMATIC is informative. I've had pretty good luck with simply inserting series resistors (say, 40Ω) near the SD card socket. You can get ganged resistors to make that easier.

callous4567, 2023-01-23:
Aye- it likely is electrical as you say. Given that it worked fine at 44 MHz I am just assuming 3V3 signalling is enabled fine- otherwise it wouldn't want anything past 25 MHz (my understanding at least...) Sadly I don't have an oscilloscope- just a cheap logic analyzer that caps out at 12 MHz 😅. 

My current PCB doesn't have any sort of termination sadly- it's my first PCB... has a big fat bit of perfboarded stuff on the back because of some mistakes I made on it! The next version (3rd, technically- I skipped wasting parts on #2 since it took so damn long to arrive and they add up to £50-ish) is currently sitting in Hong Kong celebrating lunar new year 😐. 

It has some 0605 pads for source termination on it as a lot of folk (including you now!) have recommended... hopefully that makes the difference- I'll make sure to report back on whether it fixes the issue (after lunar new year I suppose xD) ... I'll also look into that SMD you mentioned (or something similar)... might save the effort of soldering half a dozen pullups and a bunch of ferrites!

callous4567, 2023-01-26:
Edit: the addition of termination didn't get it up to 50 MHz sadly... I won't be able to experiment with this more without a scope to see whether the signals get through well enough. The exact mode of failure is (specifically)

`sd_read_block: Invalid CRC received 0xffff result of computation 0xded0
`

I'll just stick to 30-40 MHz!

carlk3, 2023-01-26:
Did you try messing with the drive strength? I've found that, paradoxically, sometimes lower is better. The very fast slew rate of the higher drive strengths is more likely to cause overshoots and reflections and stuff. Of course, to run at that speed you do need a darn fast slew rate. I don't know if there are enough numbers in the data sheets that you could figure it out analytically.

carlk3, 2023-01-30:
> Aye, SDIO would be fantastic ^_^
I've added support for 4-bit wide SDIO in a new branch:
https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio

callous4567, 2023-01-31:
> Did you try messing with the drive strength? I've found that, paradoxically, sometimes lower is better. The very fast slew rate of the higher drive strengths is more likely to cause overshoots and [...]

Aye I set it to the min and the max- neither fixed it. At this point I have most of my other code working fine and I only really need 1 MBps write in the worst-case-scenario so I'm safe enough running at 25 MHz. 50 would have been cool though ;-; 

> > Aye, SDIO would be fantastic ^_^
> > I've added support for 4-bit wide SDIO in a new branch:
> > https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio

Thanks! :D My next project will be more image-based (and so might need higher write speeds) and I'll keep the SDIO functionality in mind for that :p