carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #42: Semaphore use in spi.c

3s1d, 2023-01-12:
Hi,

I got a bit confused by the use of 
'''
sem_release(&pSPI->sem);
'''
in void spi_irq_handler(spi_t *pSPI) in file spi.c.

First of all I am using FreeRTOS in SMP mode. Releasing a semaphore inside of an IRQ will freeze everything. So I moved it into spi_transfere() line 112, where I even got more confused. You are acquiring the semaphore for the spi AFTER starting it. I mean shouldn't this be done prior to it? That got me thinking why you are using it in the first place. OK maybe for shared SPI systems. But more important what I'd like to point out is that you could release the semaphore in line 112 and get rid of the IRQ handler completely as the busy flag is cleared automatically. 
I ended up removing both, semaphore and irq handler. Works great for me.

Many thanks!
Juergen

carlk3, 2023-01-14:
Hi Juergen,

> First of all I am using FreeRTOS in SMP mode. 

Are you aware that I have a separate project for FreeRTOS in SMP mode? Please see [FreeRTOS-FAT-CLI-for-RPi-Pico](https://github.com/carlk3/FreeRTOS-FAT-CLI-for-RPi-Pico).

> Releasing a semaphore inside of an IRQ will freeze everything. 

I don't understand that. Could you please elaborate?

> So I moved it into spi_transfere() line 112, where I even got more confused. You are acquiring the semaphore for the spi AFTER starting it. 

That semaphore is used only for deferred interrupt handling. This code was ported from a FreeRTOS project, and in FreeRTOS, you don't want to wait for an interrupt by just spinning in a task checking a flag; it's best to put the task in a Blocked state and notify it somehow to unblock it. (See [Deferred Interrupt Handling](https://www.freertos.org/deferred_interrupt_processing.html).)  FreeRTOS has a variety of mechanisms that can be used for this; e.g., Direct to Task Notifications, Queues, Mutexes, Semaphores, Event Groups, etc. Since this is the "no-OS" project, there are fewer mechanisms available. I chose to use `sem_acquire` / `sem_release`.

> I ended up removing both, semaphore and irq handler. Works great for me.

If I understand correctly, you're just assuming that the SPI transaction will complete in time for you use the results. In general, that might not always be a good assumption; for example, at lower baud rates.

     -Carl

3s1d, 2023-01-16:
Hi Carl,

Bummer! oh well... I totally missed your freeRTOS alternative. Sorry my bad.
Here the ISR makes much more sense. For the noOS version you could remove it.

As I sidenote: I reused your spi/dma driver for accessing a PSRAM @ 62.5Mhz (full speed).
It stopped working for such speeds. I blame the folloing line:
```
        static uint8_t dummy = 0xA5;
```

I assume that 'static' maybe gets casted to 'static const' and always gets stored in flash and I expect that the DMA moving data from flash to SPI at a freq of CPU/2 can not keep up. Thou' in faireness I have not checked (=dereferenced the pointers) to confirm that. All I did was:

```
	//note: static not required as the function will wait for completetion.
	uint8_t dummy = SPI_FILL_CHAR;
		
    // tx write increment is already false
    if (tx) 
	{
        channel_config_set_read_increment(&pSPI->tx_dma_cfg, true);
    } 
	else 
	{
        tx = &dummy;
        channel_config_set_read_increment(&pSPI->tx_dma_cfg, false);
    }

    // rx read increment is already false
    if (rx) 
	{
        channel_config_set_write_increment(&pSPI->rx_dma_cfg, true);
    } else 
	{
        rx = &dummy;
        channel_config_set_write_increment(&pSPI->rx_dma_cfg, false);
    }
```

As the functions waits for completion, static values are note required.
Bests,
Juergen

closed 2023-01-17 by 3s1d