carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #45: Can't get simple_example to work

Addy771, 2023-02-06:
Hello. I've been trying to get an SD card mounted for reading/writing but I haven't been able to get it working. I first tried with my own code, but since it didn't work I also tried to use simple_example but that also gave the same results.

Here is the printout I get when running simple_example:
[image]

This is with a Chinese 2GB uSD card. I've also tried an 8GB Sandisk consumer uSD and an 8GB Sandisk industrial uSD. The result is the same, other than different sizes/values being reported.

I thought signal integrity could be a problem. I've tried setting baud_rate to 25, 12.5, 10, 5, 1 and 0.4MHz with no success. I have pullups on MISO and CS and the uSD breakout board I'm using also has series resistors on the signal lines. When I checked the waveforms with my oscilloscope, they look fine. The slew rates are fast enough and ringing is minimal. 

I've also checked the signals with a logic analyzer. Here's a snippet of the low-speed communication:
[image]

Clearly the SD cards are responding to the pico, based on the data received and the logic analyzer captures. I do see something strange though, after the low-speed communication there's only a single byte of high speed at the end:
[image]

I'm not sure what else to check. I've tried having the SD card wired up to SPI0 and SPI1 but there was no difference. If you have any suggestions for me I would greatly appreciate them.

carlk3, 2023-02-06:
Those logic analyzer traces look good. I recently spent about a day trying to figure out a similar problem, and finally decided to hook up the oscilloscope just to make sure the 3.3 V was solid. And... as soon as I hooked up the oscilloscope, the test ran perfectly for the first time. Turned out to be a ground loop problem, I think. I went back and made sure that I only had one external ground, and I hooked up every ground pin on the Pico, and whatever else I could find. But, it had always looked good on the logic analyzer.

Anyway, start with the basics and make sure you have solid grounds and 3.3 V. I put a 0.33 uF tantalum decoupling cap. between the 3.3 V and GND close to the SD card socket for good luck (even though there was a 10 uF and 0.1 uF already there) (0.33 uF just being what I had on hand).

Next, I would try the [Low Level I/O Test](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/master/example/tests/app4-IO_module_function_checker.c) (originally from https://github.com/abbrev/fatfs/blob/master/documents/res/app4.c). It will destroy the format of the SD card, but it is much easier to debug at the block level than at the application level. You can use a debugger to step through the "Single sector write test", for example, and trace that on the logic analyzer. You can run the Low Level I/O Test as a command `lliot` from the [example](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/master/example) program. Only once that runs perfectly would I attempt to use the higher level file system functions.

Addy771, 2023-02-07:
Thanks for the help. I added a 100uF capacitor on the SD power and checked it with the scope. There were some dips in the supply going to to about 3.0V briefly when the card was idle, possibly switching noise from the pico DC/DC converter. I added a separate regulator to provide 3.3V to the SD card. This cleared up the supply dips, but didn't help fix the issue. 

I tried running the Low Level I/O Test:
[image]

I don't have a debugger on hand, so I guess I'll set up another pico with picoprobe. I also plan to swap out my SD card breakout with a homemade breakout to see if that makes any difference.

carlk3, 2023-02-07:
Wow, that is a strange result.  The fact that GET_SECTOR_COUNT succeeds is a good sign, because the driver has to go to the card to get that information. The fact that GET_SECTOR_SIZE is called at all is weird. That would mean that FF_MAX_SS != FF_MIN_SS. For all of the SD cards I've heard of, the sector size is fixed at 512 bytes. Those are defined in `ff15\source\ffconf.h`:
```
#define FF_MIN_SS		512
#define FF_MAX_SS		512
```

Addy771, 2023-02-07:
Oh, I forgot I had messed with that setting when trying to get things working. I'd set FF_MAX_SS to 4096. I was confused because when formatting the 8GB cards I was unable to set the "allocation unit size" as low as 512.

[image]

carlk3, 2023-02-07:
Yeah, that's not the same thing as the sector size. See [f_mkfs](http://elm-chan.org/fsw/ff/doc/mkfs.html).

Addy771, 2023-02-08:
Alright, with FF_MAX_SS restored to the default value and my DIY uSD card breakout, the problem seems to be solved.

I can run the Low Level I/O Test with passing results on each of the uSD cards I have here. I was also able to run the simple_example at last. Here's the breakout I made:
[image]

For reference, here's the breakout which would not work, as well as the uSD cards I tested:
[image]

I'm still not sure what exactly is wrong with this uSD breakout board. If signal integrity was a problem I think it should still have worked when I turned the SPI speed down below 1MHz. There's barely any circuitry on this breakout, there shouldn't be much to go wrong. Oh well, I guess I just won't use the faulty breakout board.

closed 2023-07-28 by carlk3