carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #49: Readme pinout table does not match example, changing them in hw_config.c did not work

rfebbo, 2023-02-16:
Hello,

I noticed the pins in hw_config.c don't match the pins in the table in the readme. I thought this was not a big deal since I could just change the pins to match the table (I had already wired it up like that) then recompile. However when I did this I would get lots of these messages:

```
/home/rfebbo/git/no-OS-FatFS/FatFs_SPI/sd_driver/sd_card.c:462t
sd_wait_ready failed  
```

I thought there was something wrong with the SD card loader, but when I checked the pins with a scope there was no clock or anything. 

I was sure to triple check the pin numbers and that they were matching the GPIO pinout etc...

I reverted the pins back to how the example was originally and rewired the SD card loader to match and recompiled and it works now.

Am I missing something when it comes to changing the pin numbers? 

Thanks

carlk3, 2023-02-16:
Hi,
The hardware configuration has to be part of the application, not the library. So, any of the `hw_config.c` files you see here are just an example.  (In the beginning, I wasn't very clear about this, and the hardware configuration was more coupled to the library.) For a while, I tried to preserve the example files, but this became untenable because I use several different hardware configurations, and there have been changes to the hardware configuration specification (e.g., the addition of the [new SDIO support](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio), which required incompatible changes), and it became too time consuming to test all of the combinations. 

If you look at the new [sdio](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio) branch, I have given up on putting wiring tables in the README. I put in a link to a [spreadsheet](https://docs.google.com/spreadsheets/d/1BrzLWTyifongf_VQCc2IpJqXWtsrjmG7KnIbSBy-CPU/edit#gid=1596756903), instead. (Also, [PlatformIO](https://registry.platformio.org/libraries/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico)'s markdown interpreter seems to deal poorly with tables.) I suggest you make a spreadsheet of your own to keep track of your hardware configuration.

I would recommend using the examples as starting points that you can copy and make into your own applications. They will typically need to be customized for your hardware configuration. Just look at all of the boards that are out there off the shelf today (a small [sample](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio#notes-about-prewired-boards-with-sd-card-sockets))). They all seem to use different wiring. 

In the new branch, I have greatly expanded the [Customizing for the Hardware Configuration](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio#customizing-for-the-hardware-configuration) section.

carlk3, 2023-02-16:
I would recommend making a separate copy of the example with a customized `hw_config.c` . No matter what's in there, it will always be specific to a board. You can `diff` the example files to see if there's anything new compared to your copy. 

I do make an effort not to introduce any "breaking" changes to the library itself. That's why I made a separate branch for SDIO support: it requires incompatible changes to the layout of `hw_config.c`. From now on, I plan to pretty much leave the "master" branch alone. 

Old versions of `hw_config.c` are still around [in GitHub](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/commits/master/example/hw_config.c). I'll have to poke around and see if I can find one that matches the table in the old README and put that numbering back. The trouble is, I have no easy way to test that, since I've moved on to other hardware (the original solderless breadboard was just too unreliable). 

I'm also experimenting with the GitHub "Release" feature. I made one yesterday, intending that to be only for the `sdio` branch, but I screwed it up and it applied to everything. Once I get that figured out, hopefully that will provide a mechanism to clearly delineate the changes over time and provide an easy way to get a particular level that will never change.

rfebbo, 2023-02-16:
OK, thanks for your help and the amazing library!

closed 2023-02-16 by rfebbo

carlk3, 2023-02-16:
> I thought there was something wrong with the SD card loader, but when I checked the pins with a scope there was no clock or anything.

I'm just looking at the old configuration at https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/d8f42ede3c0aaf54f86a4ddfea161963c9ccf2b7/example/hw_config.c and I see that it was using `spi0`, whereas the [SparkFun RP2040 Thing Plus](https://learn.sparkfun.com/tutorials/rp2040-thing-plus-hookup-guide/hardware-overview)-like setup that's there now uses `spi1`, which might be why you didn't see anything on the pins. 

I don't like those wirings, because they don't work for SDIO. At https://docs.google.com/spreadsheets/d/e/2PACX-1vRtoSLP8MF7FNAsoi6Fp5NZoT13bsbu7kOO_5kroh9spW2HBjNhZlKBhSyBi026rd0RkEqgOCxp64cG/pubhtml?gid=990746506&single=true there is at least one wiring that can work as either SPI or SDIO. I'm toying with the idea of going to [JLCPCB](https://jlcpcb.com/) and making a custom RP2040 SD Card Development Board that has a dual-purpose SPI/SDIO socket, a dedicated SPI channel with two SD card sockets, and a dedicated SDIO bus with two SD card sockets, along with a PicoProbe. (Sockets are, like, $0.42.) Then, I should be able to easily test many configurations. (Getting two SD cards working on a shared SDIO bus will be a new challenge.)

carlk3, 2023-02-17:
I have put the old pin numbers back in `example/hw_config.c` in the `dev` branch, to agree with the README. You can `git checkout dev` and give that a try. If it looks good, I'll merge it with the `master` branch.