carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #52: Example can't work

eugnio1, 2023-02-19:
I've modified hw_config.c, thing_plus.hw_config.c, CMakeLists.txt for usb instead of UART and build successfully, but nothing is in the sd card
hw:
`/* hw_config.c
Copyright 2021 Carl John Kugler III

Licensed under the Apache License, Version 2.0 (the License); you may not use
this file except in compliance with the License. You may obtain a copy of the
License at

   http://www.apache.org/licenses/LICENSE-2.0
under the License is distributed on an AS IS BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND, either express or implied. See the License for the
specific language governing permissions and limitations under the License.
*/
/*

This file should be tailored to match the hardware design.

There should be one element of the spi[] array for each hardware SPI used.

There should be one element of the sd_cards[] array for each SD card slot.
The name is should correspond to the FatFs "logical drive" identifier.
(See http://elm-chan.org/fsw/ff/doc/filename.html#vol)
The rest of the constants will depend on the type of
socket, which SPI it is driven by, and how it is wired.

*/

#include <string.h>
//
#include "my_debug.h"
//
#include "hw_config.h"
//
#include "ff.h" /* Obtains integer types */
//
#include "diskio.h" /* Declarations of disk functions */

// Hardware Configuration of SPI "objects"
// Note: multiple SD cards can be driven by one SPI if they use different slave
// selects.
static spi_t spis[] = {  // One for each SPI.
    {
        .hw_inst = spi0,  // SPI component
        .miso_gpio = 16,  // GPIO number (not Pico pin number)
        .mosi_gpio = 19,
        .sck_gpio = 18,
        .set_drive_strength = true,
        .mosi_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,
        .sck_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,

         .baud_rate = 1000 * 1000,
        //.baud_rate = 12500 * 1000,  // The limitation here is SPI slew rate.
//        .baud_rate = 25 * 1000 * 1000, // Actual frequency: 20833333. Has
        // worked for me with SanDisk.
    }
};

// Hardware Configuration of the SD Card "objects"
static sd_card_t sd_cards[] = {  // One for each SD card
    {
        .pcName = "0:",   // Name used to mount device
        .spi = &spis[0],  // Pointer to the SPI driving this card
        .ss_gpio = 17,     // The SPI slave select GPIO for this SD card
        .set_drive_strength = true,
        .ss_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,
        .use_card_detect = true,
        .card_detect_gpio = 13,   // Card detect
        .card_detected_true = -1   // What the GPIO read returns when a card is
                                  // present.
    }
};

/* ********************************************************************** */
size_t sd_get_num() { return count_of(sd_cards); }
sd_card_t *sd_get_by_num(size_t num) {
    if (num <= sd_get_num()) {
        return &sd_cards[num];
    } else {
        return NULL;
    }
}
size_t spi_get_num() { return count_of(spis); }
spi_t *spi_get_by_num(size_t num) {
    if (num <= sd_get_num()) {
        return &spis[num];
    } else {
        return NULL;
    }
}

/* [] END OF FILE */`
thpls.hw:
`/* hw_config.c
Copyright 2021 Carl John Kugler III

Licensed under the Apache License, Version 2.0 (the License); you may not use
this file except in compliance with the License. You may obtain a copy of the
License at

   http://www.apache.org/licenses/LICENSE-2.0
under the License is distributed on an AS IS BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND, either express or implied. See the License for the
specific language governing permissions and limitations under the License.
*/
/*

This file should be tailored to match the hardware design.

There should be one element of the spi[] array for each hardware SPI used.

There should be one element of the sd_cards[] array for each SD card slot.
The name is should correspond to the FatFs "logical drive" identifier.
(See http://elm-chan.org/fsw/ff/doc/filename.html#vol)
The rest of the constants will depend on the type of
socket, which SPI it is driven by, and how it is wired.

*/

#include <string.h>
//
#include "my_debug.h"
//
#include "hw_config.h"
//
#include "ff.h" /* Obtains integer types */
//
#include "diskio.h" /* Declarations of disk functions */

// Hardware Configuration of SPI "objects"
// Note: multiple SD cards can be driven by one SPI if they use different slave
// selects.
static spi_t spis[] = {  // One for each SPI.
    {
        .hw_inst = spi1,  // SPI component
        .miso_gpio = 16,  // GPIO number (not pin number)
        .mosi_gpio = 19,
        .sck_gpio = 18,
        .set_drive_strength = true,
        .mosi_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,
        .sck_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA,

         .baud_rate = 1000 * 1000,
        //.baud_rate = 12500 * 1000,
//        .baud_rate = 25 * 1000 * 1000,  // Actual frequency: 20833333.
    }};

// Hardware Configuration of the SD Card "objects"
static sd_card_t sd_cards[] = {  // One for each SD card
    {
        .pcName = "0:",           // Name used to mount device
        .spi = &spis[0],          // Pointer to the SPI driving this card
        .set_drive_strength = true,
        .ss_gpio_drive_strength = GPIO_DRIVE_STRENGTH_2MA
    }
};

/* ********************************************************************** */
size_t sd_get_num() { return count_of(sd_cards); }
sd_card_t *sd_get_by_num(size_t num) {
    if (num <= sd_get_num()) {
        return &sd_cards[num];
    } else {
        return NULL;
    }
}
size_t spi_get_num() { return count_of(spis); }
spi_t *spi_get_by_num(size_t num) {
    if (num <= sd_get_num()) {
        return &spis[num];
    } else {
        return NULL;
    }
}

/* [] END OF FILE */`
[image]

eugnio1, 2023-02-19:
I've put the code in brackets why it isn't right

carlk3, 2023-02-19:
> I've put the code in brackets why it isn't right

To be honest, I don't see the brackets.

Anyway, I took another stab at it, and pushed new versions of `example/hw_config.c` and `example/thing_plus.hw_config.c` to the `dev` branch. I think these agree with the tables in the README and the [RP2040 Thing Plus Hookup Guide](https://learn.sparkfun.com/tutorials/rp2040-thing-plus-hookup-guide/hardware-overview). I added some comments.

> I've modified ... thing_plus.hw_config.c, 

You only need to use that if you have a [SparkFun Thing Plus - RP2040](https://www.sparkfun.com/products/17745). You would have to rename it to `hw_config.c` (or change the `CMakeLists.txt`).

> CMakeLists.txt for usb instead of UART 

What did you have to change? Is the USB terminal working? Are there any messages on the terminal?

carlk3, 2023-02-20:
I'd suggest going back to "Chapter 4. Saying "Hello World" in C" in [Getting started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf). Get that working first, before you try anything more complicated.

carlk3, 2023-02-21:
I 'd suggest comparing the `CMakeLists.txt files` to see what might be causing the difference in behavior. 

You can always modify `FatFS_SPI_example.cpp` `main` to `printf` something you would recognize. Normally, it will just print "> " as a prompt.

eugnio1, 2023-02-22:
Oh! I didn't know that I have to press enter in the first place for " > " to appear , never thought of it that way. Any possible solutions for "No SD card detected!
f_getcwd error: The physical drive cannot work (3)"
[image]

carlk3, 2023-02-22:
Just a guess, but from your photo it looks like you might be using an Arduino SD card board. Those are for 5 V Arduinos and _will _not__ work on 3.3 V Raspberry Pi Pico.

eugnio1, 2023-02-22:
Thanks for your time. Changing .card_detected_true = -1 doesn't do the job.

closed 2023-02-22 by eugnio1

carlk3, 2023-02-23:
You need 
```
    .use_card_detect = false
```
`.card_detected_true ` is the value that `gpio_get` of the `.card_detect_gpio` returns when a card is present. For some sockets, this will be 0, for others, it will be 1.