carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #59: FatFs_SPI fails to compile with Pi Pico SDK v1.5 on Windows

NicoAICP, 2023-04-14:
Hey there, i am trying to use your library in a project of mine, which i am trying to update to 1.5 to not use linux anymore (currently i have to compile it by running a vm)

I used this to install the sdk: https://github.com/raspberrypi/pico-setup-windows/releases/latest/download/pico-setup-windows-x64-standalone.exe

When i try to compile your library tho, i get following issue:
>[main] Building folder: no-OS-FatFS-SD-SPI-RPi-Pico 
[main] Configuring project: no-OS-FatFS-SD-SPI-RPi-Pico 
[proc] Executing command: "C:\Program Files\Raspberry Pi\Pico SDK v1.5.0\cmake\bin\cmake.EXE" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -SC:/Users/jfiop/Documents/Pico-v1.5.0/no-OS-FatFS-SD-SPI-RPi-Pico/FatFs_SPI -Bc:/Users/jfiop/Documents/Pico-v1.5.0/no-OS-FatFS-SD-SPI-RPi-Pico/build -G "NMake Makefiles"
[cmake] Not searching for unused variables given on the command line.
[cmake] CMake Warning (dev) in CMakeLists.txt:
[cmake]   No project() command is present.  The top-level CMakeLists.txt file must
[cmake]   contain a literal, direct call to the project() command.  Add a line of
[cmake]   code such as
[cmake] 
[cmake]     project(ProjectName)
[cmake] 
[cmake]   near the top of the file, but after cmake_minimum_required().
[cmake] 
[cmake]   CMake is pretending there is a "project(Project)" command on the first
[cmake]   line.
[cmake] This warning is for project developers.  Use -Wno-dev to suppress it.
[cmake] 
[cmake] -- The C compiler identification is MSVC 19.29.30145.0
[cmake] -- The CXX compiler identification is MSVC 19.29.30145.0
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - failed
[cmake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
[cmake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - broken
[cmake] CMake Error at C:/Program Files/Raspberry Pi/Pico SDK v1.5.0/cmake/share/cmake-3.25/Modules/CMakeTestCCompiler.cmake:70 (message):
[cmake]   The C compiler
[cmake] 
[cmake]     "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"
[cmake] 
[cmake]   is not able to compile a simple test program.
[cmake] 
[cmake]   It fails with the following output:
[cmake] 
[cmake]     Change Dir: C:/Users/jfiop/Documents/Pico-v1.5.0/no-OS-FatFS-SD-SPI-RPi-Pico/build/CMakeFiles/CMakeScratch/TryCompile-c35gat

Is the library not supposed to be compiled on Windows? I can compile it on Linux just fine.

carlk3, 2023-04-16:
That directory ("Building folder: no-OS-FatFS-SD-SPI-RPi-Pico") isn't meant to be compiled. Is there even a CMakeLists.txt? Generally, each application would have its own CMakeLists.txt which would have an `add_subdirectory(`...`FatFs_SPI build)` to pick up the library.  For example, see [example](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/master/example).

Freyte, 2023-05-23:
Also I've been doing as you said carlk3 and with the CMake i don't have problem,but with the make it seems that does not recognize the libraries.
[image]
[image]

carlk3, 2023-05-23:
Works for me:
```
carlk@HP:/mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio/examples/simple/build$ uname -a
Linux HP 5.10.102.1-microsoft-standard-WSL2 #1 SMP Wed Mar 2 00:30:59 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
carlk@HP:/mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio/examples/simple$ rm -fr build
carlk@HP:/mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio/examples/simple$ mkdir build
carlk@HP:/mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio/examples/simple$ cd build
carlk@HP:/mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio/examples/simple/build$ cmake ..
Using PICO_SDK_PATH from environment ('/mnt/c/Users/carlk/pi/pico/pico-sdk')
PICO_SDK_PATH is /mnt/c/Users/carlk/pi/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /mnt/c/Users/carlk/pi/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /home/carlk/.platformio/penv/bin/python3.8 (found version "3.8.10") found components: Interpreter
TinyUSB available at /mnt/c/Users/carlk/pi/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /mnt/c/Users/carlk/pi/pico/pico-sdk/lib/btstack
cyw43-driver available at /mnt/c/Users/carlk/pi/pico/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /mnt/c/Users/carlk/pi/pico/pico-sdk/lib/lwip
mbedtls available at /mnt/c/Users/carlk/pi/pico/pico-sdk/lib/mbedtls
Using PICO_EXAMPLES_PATH from environment ('/mnt/c/Users/carlk/pi/pico/pico-examples')
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio/examples/simple/build
carlk@HP:/mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio/examples/simple/build$
```
Maybe you need to clear out your CMake cache?

What is your directory structure? Mine looks like this:
```
carlk@HP:/mnt/c/Users/carlk/pi/pico/no-OS-FatFS-SD-SPI-RPi-Pico.sdio$ tree
.
├── LICENSE
├── README.md
├── examples
│   ├── PlatformIO
│   │   ├── bench2
│   │   │   ├── include
│   │   │   │   └── README
│   │   │   ├── lib
│   │   │   │   └── README
│   │   │   ├── platformio.ini
│   │   │   ├── src
│   │   │   │   └── main.cpp
│   │   │   └── test
│   │   │       └── README
│   │   ├── data_logger
│   │   │   ├── platformio.ini
│   │   │   └── src
│   │   │       └── main.cpp
[... 115 lines cut ...]
    │   │   └── sd_card_sdio.c
    │   ├── SPI
    │   │   ├── crc.c
    │   │   ├── crc.h
    │   │   ├── sd_card_constants.h
    │   │   ├── sd_card_spi.c
    │   │   ├── sd_card_spi.h
    │   │   ├── sd_spi.c
    │   │   ├── sd_spi.h
    │   │   ├── spi.c
    │   │   └── spi.h
    │   ├── SdCardInfo.h
    │   ├── hw_config.h
    │   ├── sd_card.c
    │   └── sd_card.h
    └── source
        ├── FatFsSd.cpp
        ├── f_util.c
        ├── ff_stdio.c
        ├── glue.c
        ├── my_debug.c
        └── rtc.c

40 directories, 115 files
```

Freyte, 2023-05-24:
I added it to an another project, i check the directories and files and everything is okay. If i make the simple example it works, but when i add to the other project and CMake, it seems not read it.
I use add_subdirecty () to link them.
[image]
[image]
[image]

closed 2023-07-28 by carlk3