carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #61: Set irqShared from define

Mendiax, 2023-04-30:
Hi, I am using this repo in my project as a submodule. I want to set irqShared to false in spi.c, but it would be good to change it with define without modifying the source file.

Now it is:
static bool irqShared = true;

But it could be done like this:
#ifndef NO_OS_FATFS_IRQ_SHARED
#define NO_OS_FATFS_IRQ_SHARED true
#endif
static bool irqShared = NO_OS_FATFS_IRQ_SHARED;

Doing it this way it is possible to 
add to cmake add_definitions(-DNO_OS_FATFS_IRQ_SHARED=0) line that will change it to not shared mode without modifying source code.

carlk3, 2023-05-02:
Why do you need no-OS-FatFS-SD-SPI-RPi-Pico to use an exclusive interrupt?

carlk3, 2023-07-28:
Not sure why you would need to do this, but if you want to make a Pull Request, I'll merge it.

Mendiax, 2023-08-01:
I have issues with mixing shared and exclusive irq handlers, but I found PICO_DISABLE_SHARED_IRQ_HANDLERS define build in pico lib that fixes my problem. It just make all shared handlers exclusive handlers, so it is better than sollution I proposed.

carlk3, 2023-08-01:
> I have issues with mixing shared and exclusive irq handlers

Well, that's concerning. Also, I saw this post about reasons to avoid shared irq handlers:
https://forums.raspberrypi.com/viewtopic.php?t=354129&sid=4fd9f42c13e6de2e81d975643b38ac0d#p2122816

The thing is, with only two DMA system IRQs in the whole RP2040, one runs out pretty quickly when one starts using PIO and stuff if they are used exclusively. As a library writer, I pretty much have to deal with shared irq handlers, so I would hope that they can be made to work! I agree that the SDK implementation and documentation is not very good. I don't have great confidence in my use of them.

> I found PICO_DISABLE_SHARED_IRQ_HANDLERS define build in pico lib
Good to know!