carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #62: SDIO example?

piverson110, 2023-05-10:
As far as I can see, there is no example of SDIO being implemented in this library. I see "Customising the hardware configuration" and some relevant structures, but the documentation and examples are focused on SPI. I know that the specific configuration will depend on how the sd card is wired, but does anyone have a working SDIO example to share?

carlk3, 2023-05-11:
Are you looking at the [sdio ](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio) branch?

[dynamic_config](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio/examples/dynamic_config) uses SDIO. There are also a few examples in the [PlatformIO](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio/examples/PlatformIO) subdirectory that use SDIO.

piverson110, 2023-05-11:
Cheers, I'll check those out.

carlk3, 2023-05-12:
Also, the [command_line](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio/examples/command_line) example shows SDIO in a static configuration (with [hw_config.c](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/sdio/examples/command_line/hw_config.c)).

piverson110, 2023-05-15:
Thanks! I got confused by the various options, but now I've got it working easily on a self-designed PCB.

closed 2023-05-15 by piverson110