carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #64: Question.?

Freyte, 2023-05-15:
is it a version for Micro Sd card Adapter in C ,

carlk3, 2023-05-15:
Yes. One C example: [simple](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio/examples/simple)

Freyte, 2023-05-16:
Really thank you, i tried to work with but I am having a problem to made the .uf2 of the simple. Did you also have this kind of problems or similar? 
[image]

carlk3, 2023-05-17:
What `gcc` version are you using? I'm using

`arm-none-eabi-gcc.exe (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)`

I don't know how long `static_assert` has been around. Anyway, it is just a compile time assertion.. You can just comment it out.

Freyte, 2023-05-17:
I have a version gcc (Debian 10.2.1-6) 10.2.1 Copyright (c) 2020 Free Software Foundation, Inc.
I took your advice to comment it out and works. Thank you so much

closed 2023-05-17 by Freyte