carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #70: Bytes consistently skipped when reading from one file

schollz, 2023-07-21:
This library is absolutely amazing, thank you soooo much for maintaining it. I don't expect any sort of answer here, but I just wanted to throw out my situation in case anyone has any hint of how to trouble shoot a weird issue.

I put a file "`3.raw`" (audio data) onto the sd card and read it back using the library but it consistently gives back *some* of the bytes from the file, in a consistent way, but *never* all of the bytes from the file. What reasons could this be occurring? [Here is the specific code that reads the bytes](https://gist.github.com/schollz/67445a165b324d354f432ebab3307a2d#file-main-c-L131-L145).

Here's what I've troubleshooted so far:

- I can use this library to write+read a file and it works perfectly. I can write "`This is another test of writing to an SD card.`" and read that back completely in a file. This indicates to me that the SD card wiring, the SD card itself, the code are functioning (code attached below).
- I can use this library to get the file info of the offending file (`3.raw`) which also works perfectly and gives the correct file size.
- I copied the file `3.raw` several times onto the SD card and tried reading from those files and each one is read in the exact same way, with some bytes skipped.
- Here is what the byte skipping looks like:

[image]

The `xxd` output is from the file *on the SD card*. I'm reading bytes using a buffer of 100 bytes on the pico and only retrieving the bytes in blue. The bytes in yellow never show up in the output. The bytes that *are* read do appear in the correct order.

<details><summary>This is the actual output of the code running on the pico (code below)</summary>

```
sd_spi_go_low_frequency: Actual frequency: 398089
V2-Version Card
R3/R7: 0x1aa
R3/R7: 0xff8000
R3/R7: 0xc0ff8000
Card Initialized: High Capacity Card
SD card initialized
SDHC/SDXC Card: hc_c_size: 29519
Sectors: 30228480
Capacity:    14760 MB
sd_spi_go_high_frequency: Actual frequency: 992063
hello.txt: fsize: 49
Reading from file 'hello.txt':
---                                                                                     
This is another test                                                                    
of writing to an SD card.                                                               
                                                                                        
---                                                                                     
                                                                                        
                                                                                        
Reading from file 'hello.txt':                                                          
s is another test                                                                       
of writing to an SD card.                                                               
3.raw: fsize: 498072                                                                    
                                                                                        
                                                                                        
Reading from file '3.raw':                                                              
read: 23744 bytes                                                                       
                                                                                        
                                                                                        
Reading 32 bytes from file '3.raw':                                                     
61                                                                                      
31                                                                                      
6c                                                                                      
2f                                                                                      
32                                                                                      
55                                                                                      
da                                                                                      
ac                                                                                      
4                                                                                       
70                                                                                      
37                                                                                      
37                                                                                      
58                                                                                      
60                                                                                      
2a                                                                                      
68                                                                                      
18                                                                                      
11                                                                                      
1                                                                                       
53                                                                                      
24                                                                                      
a                                                                                       
0                                                                                       
1                                                                                       
0                                                                                       
1                                                                                       
0                                                                                       
1                                                                                       
0                                                                                       
1                                                                                       
0                                                                                       
0                                         
```

</details>

More details:

- Here is my code, `hw_config.c` and CMakeLists.txt: https://gist.github.com/schollz/67445a165b324d354f432ebab3307a2d. And [here is the specific code that reads the bytes](https://gist.github.com/schollz/67445a165b324d354f432ebab3307a2d#file-main-c-L131-L145).
- The SD card is Kingston Micro SD HC 16 GB
- I'm using a Pico connected [to this micro sd card adapter](https://www.amazon.com/gp/product/B07PFDFPPC)

Anyways, I appreciate ANY help that can be provided. I'm totally stumped. I'm used to when code simply doesn't work, but in this case its working 100% for writing/reading text, but for whatever reason only working 30% for reading from a binary file.

schollz, 2023-07-21:
Update: I'm using the [SDIO branch](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/tree/sdio#customizing-for-the-hardware-configuration) now and compiled and ran the `command_line` example which works fantastic. I can mount the drive and cat files. I edited the cat file to output bytes, but again I can see that its outputting bytes in the same consistent manner as before, skipping bytes exactly how I outlined in the previous post.

Since the code is working, both SD card readers are working, the pico is working, I'm assuming it must be the SD card itself?

carlk3, 2023-07-23:
If you've got the `command_line` example running, have you run some of the tests? For example, `big_file_test` can write gigabytes of pseudo-random data and then read it back, verifying each byte.

It looks like you are using [f_gets](http://elm-chan.org/fsw/ff/doc/gets.html) to read the data, am I wrong? I would not use `f_gets` to read binary data. It is intended to read strings, and will assume something like UTF-8 encoding. I would use [f_read](http://elm-chan.org/fsw/ff/doc/read.html) instead. See [big_file_test.c](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/sdio/examples/command_line/tests/big_file_test.c) for an example.

schollz, 2023-07-23:
> It looks like you are using [f_gets](http://elm-chan.org/fsw/ff/doc/gets.html) to read the data, am I wrong?

Yep, you are right - I am using `f_gets` and I am totally wrong for using it!!

> I would use [f_read](http://elm-chan.org/fsw/ff/doc/read.html) instead.

Doing this fixes everything. I am reading byte-for-byte perfectly.

Thank you so much for solving this for me @carlk3 ! Sometimes I forget to check the API, this is a really good reminder.

closed 2023-07-23 by schollz