carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #71: Unable to write to sd card

BenZonneveld, 2023-07-21:
I have a waveshare Eval board which has the CLK pin for the sd card on pin 5.
I've made some modifications to the code and am able to read from the sd card. Writing is not working though.
Do you have any pointers on what I might have missed ?

BenZonneveld, 2023-07-21:
The error I get is: f_mkfs error: A hard error occurred in the low level disk I/O layer (1)

BenZonneveld, 2023-07-25:
Forgot to mention this is on the SDIO branch.

BenZonneveld, 2023-07-26:
Looking at the schematics for the Waveshare Eval board I noticed there is no pull up resistor on the CLK line.

carlk3, 2023-07-26:
> I've made some modifications to the code

What changes did you make? Since SD_D0 is on GP19, and SD_CLK is on GP5, the `SDIO_CLK_PIN_D0_OFFSET` would have to be -14, or **18** in modulo 32 arithmetic. That change would have to be made in the PIO file: `sd_driver\SDIO\rp2040_sdio.pio`. 

I don't know how much I trust the `cmake` handling of the PIO files. Make sure the change is reflected in `sd_driver\SDIO\rp2040_sdio.pio.h`. 

Is that one of the changes you made?

> Looking at the schematics for the Waveshare Eval board I noticed there is no pull up resistor on the CLK line.

There should be no need for a pull up there.

BenZonneveld, 2023-07-27:
That was indeed the line I changed. After getting a new clone of the repo and remembering to switch to the sdio branch AND removing the sd_driver\SDIO\rp2040_sdio.pio.h (as it is regenerated anyway...) everything is working as intended!

closed 2023-07-27 by BenZonneveld