carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #72: SPI on lowest clock (20 mhz)

niansa, 2023-07-25:
Hey!

Now that I got my sd card board to work, I'd like to run the Pico at the lowest possible clock (20 mhz) (for battery saving).
However, the SD card fails to initialize when I do so.

I assume some clock divider needs to be adjusted somewhere? Can you point me to it?

Thanks
niansa

carlk3, 2023-07-25:
For initialization, the SPI clock is supposed to be in the frequency range 100 kHz - 400 kHz. This is hardcoded in [sd_spi_go_low_frequency](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/master/FatFs_SPI/sd_driver/sd_spi.c#L34).

I don't know whether [spi_set_baudrate](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#ga37f4c04ce4165ac8c129226336a0b66c) takes the system clock frequency into account.

niansa, 2023-07-27:
Thanks!
So just calling `sd_spi_go_high_frequency` when changing core frequency after SD card init is done should solve it without sideeffects?

carlk3, 2023-07-27:
I believe so, as long as you're using the SDK to change the system clock frequency.

niansa, 2023-07-28:
Well, I finally got to test it. I was fully convinced it'd work, but it actually doesn't. :-/

carlk3, 2023-08-30:
> Well, I finally got to test it. I was fully convinced it'd work, but it actually doesn't. :-/

Which branch are you running? One way or another, you can get the SPI clock's actual frequency printed out. See `FatFs_SPI\sd_driver\sd_spi.c` or `src\sd_driver\SPI\sd_spi.c`:
```
void sd_spi_go_high_frequency(sd_card_t *sd_card_p) {
    uint actual = spi_set_baudrate(sd_card_p->spi_if.spi->hw_inst, sd_card_p->spi_if.spi->baud_rate);
    DBG_PRINTF("%s: Actual frequency: %lu\n", __FUNCTION__, (long)actual);
}
```
The RP2040 has some limitations on the SPI clock frequencies it can generate, so the SDK sets the SPI frequency as close as possible to the baud rate requested, and returns the actual achieved rate. Anything from 100 kHz to 400 kHz is within spec. at initialization time, then it can go up to something like 25 MHz (but you need good hardware to reach 25 MHz). 

At what point does it fail? Does it make it through initialization (which is supposed to be at 400 kHz)?

carlk3, 2023-09-06:
I've got it running reliably at 20000 kHz. Two things:

1. The SPI is clocked by the Peripheral Clock (clk_peri). Since clk_peri is typically clocked from clk_sys, if you change clk_sys you need to reconfigure clk_peri. E.g.;
```
clock_configure(clk_peri,
                    0,
                    CLOCKS_CLK_PERI_CTRL_AUXSRC_VALUE_CLK_SYS,
                    clock_get_hz(clk_sys),
                    clock_get_hz(clk_sys));
```
2. I had to reduce the SPI baud rate to 1 MHz or I got strange DMA hangs. E.g.:
```
static spi_t spis[] = {  // One for each SPI.
    {
        .hw_inst = spi0,  // SPI component
        // ...
        .baud_rate = 1E6,  
        // ...
```
I'm not sure what's going on with the DMA at the lower clk_sys. If I try to run at baud rate 5 MHz, it hangs with 
```
RX DMA control regs:
(01088821) ber 0 rer 0 wer 0 busy 1 trq 17 cto 1 rgsl 0 rgsz 0 inw 1 inr 0 sz 1 hip 0 en 1
```
I don't get the DMA block completion interrupt, and the DMA says it's busy, but I don't see any error flags. I guess it just can't keep up with the SPI. Or, maybe the SPI can't keep up.

carlk3, 2023-09-07:
I made a few tweaks, and now I've got SPI working with a baud rate of 5 MHz with system clock at 20 MHz. I'm not really sure what made the difference.
```
pll_sys  = 20001kHz
pll_usb  = 48000kHz
rosc     = 4478kHz
clk_sys  = 20001kHz     reported  = 20000kHz
clk_peri = 20000kHz     reported  = 20000kHz
clk_usb  = 48000kHz     reported  = 48000kHz
clk_adc  = 48000kHz     reported  = 48000kHz
clk_rtc  = 46kHz        reported  = 46kHz

> mount 0:
sd_spi_go_low_frequency: Actual frequency: 400000
V2-Version Card
R3/R7: 0x1aa
R3/R7: 0x40ff8000
R3/R7: 0xc0ff8000
Card Initialized: High Capacity Card
SD card initialized
SDHC/SDXC Card: hc_c_size: 30386
Sectors: 31116288
Capacity:    15193 MB
sd_spi_go_high_frequency: Actual frequency: 5000000

> bench 0:
Type is FAT32
Card size: SDHC/SDXC Card: hc_c_size: 30386
Sectors: 31116288
Capacity:    15193 MB
15.93 GB (GB = 1E9 bytes)

Manufacturer ID: 0x3
OEM ID: SD
Product: SC16G
Revision: 8.0
Serial number: 0x9d49ce1d
Manufacturing date: 1/2021

FILE_SIZE_MB = 5
BUF_SIZE = 20480
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
264.5,239219,73102,77376
268.7,124884,73043,76191

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
294.3,73072,68797,69579
294.4,72997,68805,69570

Done

> mount 1:
sd_spi_go_low_frequency: Actual frequency: 400000
V2-Version Card
R3/R7: 0x1aa
R3/R7: 0x40ff8000
R3/R7: 0xc0ff8000
Card Initialized: High Capacity Card
SD card initialized
SDHC/SDXC Card: hc_c_size: 60947
Sectors: 62410752
Capacity:    30474 MB
sd_spi_go_high_frequency: Actual frequency: 5000000

> mount 2:
sd_spi_go_low_frequency: Actual frequency: 400000
V2-Version Card
R3/R7: 0x1aa
R3/R7: 0xff8000
Illegal command CMD:41 response 0x5
R3/R7: 0xc0ff8000
Card Initialized: High Capacity Card
SD card initialized
SDHC/SDXC Card: hc_c_size: 30199
Sectors: 30924800
Capacity:    15100 MB
sd_spi_go_high_frequency: Actual frequency: 5000000

> bench 1:
Type is FAT32
Card size: SDHC/SDXC Card: hc_c_size: 60947
Sectors: 62410752
Capacity:    30474 MB
31.95 GB (GB = 1E9 bytes)

Manufacturer ID: 0x0
OEM ID:
Product: USD
Revision: 1.0
Serial number: 0x301e
Manufacturing date: 8/2022

FILE_SIZE_MB = 5
BUF_SIZE = 20480
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
255.8,88278,78360,80020
255.9,84589,78360,80008

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
243.8,87296,83492,83991
243.8,86930,83492,83987

Done
```

niansa, 2024-03-17:
Thank you, that helped a lot!

closed 2024-03-17 by niansa