carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #73: Clock Speed Requirements

willeccles, 2023-08-04:
Apologies if I missed anything in the README, I sometimes skim things a little too vigorously.

I'm looking to use this in an application where the system clock frequency is 48MHz. Are there any issues with either SPI or SDIO at this frequency? I can use either of the above in this application, with SDIO's speed being preferable (the application involves high-speed data acquisition). Will the SDIO version work with a 48MHz system clock?

carlk3, 2023-08-04:
For SPI, you can set the SPI baud rate in the hardware configuration. This library uses the SDK call `spi_set_baudrate` which uses the SDK's `clock_get_hz`	to get the current frequency of the **peripheral clock** for UART and SPI (see `src\rp2_common\hardware_spi\spi.c` in the pico-sdk). I'd have to look in the Datasheet to see what sets the peripheral clock's frequency. But, anyway, you could always scale the baud rate in the H/W config. if necessary.

For SDIO, it's on my list of things to do to make the SDIO clock rate configurable in the H/W config. It is currently hard coded in `sd_sdio_begin` in (sd_driver\SDIO\sd_card_sdio.c)[https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/sdio/src/sd_driver/SDIO/rp2040_sdio.c]. 
```
    // Increase to 25 MHz clock rate
    // Actually, clk_sys / CLKDIV (from rp2040_sdio.pio),
    // So, say, 125 MHz / 4 = 31.25 MHz (see src\sd_driver\SDIO\rp2040_sdio.pio)
    // rp2040_sdio_init(sd_card_p, 1, 0); // 31.25 MHz
    rp2040_sdio_init(sd_card_p, 1, 128); // 20.8 MHz
```
so you would have to modify that directly. When I get around to making it configurable in the H/W config., I should use a scheme like that used in `spi_set_baudrate` and take the `clk_sys` into account.

There isn't a lot of computation going on, but it's possible that a slower system clock will slow down something to the point that you might see problems. The only thing that comes to mind is the verification of the CRCs. (For example, see `sdio_verify_rx_checksums` in [rp2040_sdio.c](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/sdio/src/sd_driver/SDIO/rp2040_sdio.c)). I got a great suggestion for how to do that in hardware ( #63 ), but that's another thing on the to do list.

willeccles, 2023-08-07:
Thanks for the info! I will look into this. Modifying it is certainly not an issue.

carlk3, 2023-09-06:
I have verified that it works reliably at system clock frequency is 48MHz and SDIO baud rate 5 MHz.

willeccles, 2023-09-06:
Awesome, thank you! Does 10MHz work? I see you noted it can't be greater than 1/4 of the system clock. Either should work for my purposes, though.

carlk3, 2023-09-06:
> Awesome, thank you! Does 10MHz work? 

Setting the system clock to 10000 kHz is "not possible" according to `check_sys_clock_khz`.

> I see you noted it can't be greater than 1/4 of the system clock. Either should work for my purposes, though.

48 MHz system clock with 10 MHz baud rate works fine:

```
> set_sys_clock_48mhz

> mount 3:

> bench 3:
Type is FAT32
Card size: 31.95 GB (GB = 1E9 bytes)

Manufacturer ID: 0x0
OEM ID:
Product: USD
Revision: 1.0
Serial number: 0x302c
Manufacturing date: 8/2022

FILE_SIZE_MB = 5
BUF_SIZE = 20480
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
3196.9,13015,5826,6384
3226.4,11195,5811,6335

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
4045.4,6709,4659,5061
4051.7,6310,4638,5054

Done

> big_file_test bf 10 1
Writing...
Elapsed seconds 7.14
Transfer rate 1434 KiB/s (1468 kB/s) (11745 kb/s)
Reading...
Elapsed seconds 6.57
Transfer rate 1558 KiB/s (1596 kB/s) (12767 kb/s)

```

willeccles, 2023-09-06:
Closing as you've answered my question and implemented a convenient solution. Thanks!

closed 2023-09-06 by willeccles