carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #74: f_open may mess up cyw43_gpio_irq_handler?

thangnguyen-69, 2023-08-24:
**Problem**
Seems like the f_open may override others interrupt? I am totally not sure about this.

**Setup**
I have an "almost minimal" example attached here. But let me sum it up first.
My application will run both core:
- core 0 is for bluetooth and its setup. 
- core 1 will initialize the sd card, then open a file using `f_open` when i press a button, and close it when i press again, and do it infinitely.

**Step to reproduce**
So during the run. I try to connect using the rfcomm command from linux:
`sudo rfcomm bind /dev/rfcomm0  "your pico bluetooth address" 1`, then open that port `/dev/rfcomm0` using `minicom`

**What went wrong**
- So, if i try to connect with `minicom` right at the start, then it worked normally, i can still receive the packet through the packet handler.(It does not actually connect, but the point here is the packet handler is called normally).
- **BUT**, if before that, i press the button once or twice. Then the packet handler will not be triggered AT ALL.  Added a breakpoint at `cyw43_gpio_irq_handler`  in `cyw43_driver.c`, and noticed that normally a gpio will trigger this handler, but after `f_read`, it cannot be triggered anymore!

[multicore_example.zip](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/files/12432302/multicore_example.zip)

carlk3, 2023-08-25:
I don't have time to study the example at the moment, but you could try changing the IRQ number used by no-OS-FatFS-SD-SPI-RPi-Pico. Which branch are you running?

In the `sdio` branch, you can specify a `DMA_IRQ_num` in your `hw_config.c`. This is documented in the [README](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/sdio/README.md):

> DMA_IRQ_num Which IRQ to use for DMA. Defaults to DMA_IRQ_0. The handler is added with irq_add_shared_handler, so it's not exclusive. Set this to avoid conflicts with any exclusive DMA IRQ handlers that might be elsewhere in the system.

In the `master` branch there is a `set_spi_dma_irq_channel` call you can make. See [spi.h](https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/master/FatFs_SPI/sd_driver/spi.h).

thangnguyen-69, 2023-08-26:
> I don't have time to study the example at the moment, but you could try changing the IRQ number used by no-OS-FatFS-SD-SPI-RPi-Pico. Which branch are you running?
>
> In the `sdio` branch, you can [...]
I am using master branch, so
We set it right before the stdio_init_all right?

carlk3, 2023-08-26:
That feature was added by [MCJack123](https://github.com/MCJack123) in #14, "Add DMA IRQ channel selection", but I think you could do it any time before the first `f_mount`.

Meanwhile, I have merged in a fix to the DMA IRQ handling that was missing in the master branch. It was using a common interrupt handler for DMA_IRQ_0 and DMA_IRQ_1 but only matching on the channel number, not the IRQ number. I made a couple of other improvements to the IRQ handling while I was at it. So, you could try pulling the latest and see if that makes any difference.

thangnguyen-69, 2023-08-31:
I am terribly sorry that i ignored the comment, really. I am pulling the latest now to see if it work. Update in a moment.

thangnguyen-69, 2023-08-31:
Currently testing, but it is still not working.

thangnguyen-69, 2023-09-13:
I have a lead on the problem. 
Read the bluetooth CYW43439 datasheet yesterday, and it seems like the Bluetooth interface must pull its Chip Select Line, which is GPIO 25 in Pico W, low and only then can it write data to the Pico through the GPIO24 line. SO what i did is after every f_read, and f_open, i manually set the CS line high again, so the bluetooth device can pull the CS line down again. And it did work. ( I hope the explanation I just gave makes sense). But again it is just a workaround.
So the problem now is 

- Why do f_read and f_open operations pull GPIO 25 low? (Other operations might do the same, but I'm unsure.) If this isn't intended, it might be a bug.
- I'm having trouble pinpointing the code causing this. Could you provide some direction on where to look for this bug? If I can fix it, I'll submit a pull request.

carlk3, 2023-09-13:
What is your no-OS-FatFS-SD-SPI-RPi-Pico hardware configuration (hw_config.c)?

Maybe there's supposed to be a pull on GPIO 25, e.g. `gpio_pull_up(25);`.

thangnguyen-69, 2023-09-13:
Yeah i just found this in your code at `sd_driver/sd_spi.c`

        static void sd_spi_deselect(sd_card_t *pSD) {
            gpio_put(pSD->ss_gpio, 1);
            LED_OFF();
            /*
            MMC/SDC enables/disables the DO output in synchronising to the SCLK. This
            means there is a posibility of bus conflict with MMC/SDC and another SPI
            slave that shares an SPI bus. Therefore to make MMC/SDC release the MISO
            line, the master device needs to send a byte after the CS signal is
            deasserted.
            */
            uint8_t fill = SPI_FILL_CHAR;
            spi_write_blocking(pSD->spi->hw_inst, &fill, 1);
        }
Why the LED_OFF though?

carlk3, 2023-09-13:
Oh, yeah, that must be it. The LED was used just to show activity on the SD card. It's something familiar to those of us who remember the early PCs. I don't really use it anymore and had more or less forgotten about it. 

You should be able to fix it by simply defining `NO_PICO_LED`. Something like
```
add_compile_definitions(NO_PICO_LED)
```
in `CMakeLists.txt` should do it.

thangnguyen-69, 2023-09-13:
> Oh, yeah, that must be it. The LED was used just to show activity on the SD card. It's something familiar to those of us who remember the early PCs. I don't really use it anymore and had more or [...]

i think i will open a pull request instead (after i am done with my thesis) to make it work out of the box on Pico W. This fucked me up so bad haha. But for now, I will close the issue. 
Thank you a lot for the guidance!

closed 2023-09-13 by thangnguyen-69