carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #93: Looking for a function to retrieve used and free space on SD card prior to write.

jimtal001, 2024-04-15:
I am storing large files on an SD card using this great library (thank you), but I need to verify that there is enough space remaining before the write and be able to send remote message if storage is low.  I have not been able to locate a function in your library to retrieve this info.  If it exist, could you show me a simple example?

carlk3, 2024-04-16:
Have a look at https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/196016f525e5b9c161f2b965ddd3045a4ef87649/example/FatFS_SPI_example.cpp#L175

jimtal001, 2024-04-16:
Thank You Carl.

Another question: is there an example showing how to use: the
get_fileinfo(...)  function?
Actually, all I need is the* file size.  F*rom reviewing the ff.c file it
looks like a call to get_fileinfo is required to access file size.

Jim

carlk3, 2024-04-16:
The `ls` function populates a FILINFO  structure: https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/196016f525e5b9c161f2b965ddd3045a4ef87649/example/FatFS_SPI_example.cpp#L215

jimtal001, 2024-04-18:
Hello Carl,
is there an example which demonstrates how to delete all files stored on
the SD card?

Thanks
Jim

carlk3, 2024-04-19:
One way would be to simply format the card: https://github.com/carlk3/no-OS-FatFS-SD-SPI-RPi-Pico/blob/196016f525e5b9c161f2b965ddd3045a4ef87649/example/FatFS_SPI_example.cpp#L122