carlk3/no-OS-FatFS-SD-SPI-RPi-Pico #95: SDIO code linked even if not used.

FreddyVRetro, 2024-10-07:
Hi,

Thanks a lot for your code, Not sure if you are aware but it is in the PicoMEM :)

The PicoMEM use the SD in SPI and I just saw that "sdio_irq_handler" is linked and take 0x1d8 bytes in RAM (that is huge :( )

I could remove it simply as I don't use the code as submodule, as I needed to merge the code with USB Ker read/Write.

carlk3, 2024-10-07:
...
> Thanks a lot for your code, Not sure if you are aware but it is in the PicoMEM :)

I'm sorry, but what is PicoMEM?

> The PicoMEM use the SD in SPI and I just saw that "sdio_irq_handler" is linked and take 0x1d8 bytes in RAM (that is huge :( )

472 bytes is huge? 0.17% of the 264kB of SRAM on a Pico, or 0.09% of the 520kB of SRAM on Pico 2?
...

FreddyVRetro, 2024-10-07:
Ok :)
Your name is on it :)
[image]

[image]

It is a Pi Pico board emulating lot of things for ISA PC.

I removed the SDIO compilation from the makefile and removed the driver init code.
Maybe I can reduce the sd definition variable as well.

carlk3, 2024-10-08:
Make sure you are compiling with `-ffunction-sections`, `-fdata-sections`, (and maybe `-Os`) and linking with `--gc-sections`.

FreddyVRetro, 2024-10-15:
Hi,

Yes, 412 Bytes for a unused code is huge. 
It really seems you did not check what the PicoMEM is doing :)
With this, I can add 4 Audio buffers for example.

I removed the SDIO code by removing the call to SDIO init in the initialisation code.